RobotLocomotion/drake #17562: Possible data race in //solvers:csdp_solver_test

mwoehlke-kitware, 2022-07-13:
Jammy CI [reports](https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy/job/linux-jammy-clang-bazel-experimental-everything-thread-sanitizer/1/consoleFull):
```
[0:23:00.206]  ==================
[0:23:00.206]  WARNING: ThreadSanitizer: data race (pid=20)
[0:23:00.206]    Write of size 8 at 0x7bb400003458 by main thread:
[0:23:00.206]      #0 memset <null> (csdp_solver_test+0x4a62e9)
[0:23:00.206]      #1 op_o external/csdp/lib/op_o.c:129:39 (csdp_solver_test+0x73bfac)
[0:23:00.206]      #2 sdp external/csdp/lib/sdp.c:644:5 (csdp_solver_test+0x73117f)
[0:23:00.206]      #3 easy_sdp external/csdp/lib/easysdp.c:470:8 (csdp_solver_test+0x72c293)
[0:23:00.206]      #4 drake::solvers::internal::csdp::cpp_easy_sdp(char const*, int, int, drake::solvers::internal::csdp::blockmatrix, double*, drake::solvers::internal::csdp::constraintmatrix*, double, drake::solvers::internal::csdp::blockmatrix*, double**, drake::solvers::internal::csdp::blockmatrix*, double*, double*) solvers/csdp_cpp_wrapper.cc:32:10 (csdp_solver_test+0x72b24f)
[0:23:00.206]      #5 drake::solvers::internal::(anonymous namespace)::SolveProgramThroughLorentzConeSlackApproach(drake::solvers::MathematicalProgram const&, drake::solvers::internal::SdpaFreeFormat const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, drake::solvers::MathematicalProgramResult*) solvers/csdp_solver.cc:371:19 (csdp_solver_test+0x718122)
[0:23:00.206]      #6 drake::solvers::CsdpSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const solvers/csdp_solver.cc:499:9 (csdp_solver_test+0x716677)
[0:23:00.206]      #7 drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const solvers/solver_base.cc:82:5 (csdp_solver_test+0xa3ff39)
[0:23:00.206]      #8 drake::solvers::test::RunSolver(drake::solvers::MathematicalProgram const&, drake::solvers::SolverInterface const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&) solvers/test/mathematical_program_test_util.cc:18:10 (csdp_solver_test+0xb1df35)
[0:23:00.206]      #9 drake::solvers::test::TestFindSpringEquilibrium::SolveAndCheckSolution(drake::solvers::SolverInterface const&, std::optional<drake::solvers::SolverOptions> const&, double) solvers/test/second_order_cone_program_examples.cc:359:7 (csdp_solver_test+0xa48525)
[0:23:00.206]      #10 drake::solvers::test::TestFindSpringEquilibrium_TestSOCP_Test::TestBody() solvers/test/csdp_solver_test.cc:309:7 (csdp_solver_test+0x50db3c)
[0:23:00.206]      #11 void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) external/gtest/googletest/src/gtest.cc:2599:10 (csdp_solver_test+0xe63f37)
[0:23:00.206]      #12 void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) external/gtest/googletest/src/gtest.cc:2635:14 (csdp_solver_test+0xe4d9e2)
[0:23:00.206]      #13 testing::Test::Run() external/gtest/googletest/src/gtest.cc:2674:5 (csdp_solver_test+0xe38890)
[0:23:00.206]      #14 testing::TestInfo::Run() external/gtest/googletest/src/gtest.cc:2853:11 (csdp_solver_test+0xe3918e)
[0:23:00.206]      #15 testing::TestSuite::Run() external/gtest/googletest/src/gtest.cc:3012:30 (csdp_solver_test+0xe399b7)
[0:23:00.206]      #16 testing::internal::UnitTestImpl::RunAllTests() external/gtest/googletest/src/gtest.cc:5870:44 (csdp_solver_test+0xe45c47)
[0:23:00.206]      #17 bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) external/gtest/googletest/src/gtest.cc:2599:10 (csdp_solver_test+0xe66d77)
[0:23:00.206]      #18 bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) external/gtest/googletest/src/gtest.cc:2635:14 (csdp_solver_test+0xe4ffb2)
[0:23:00.206]      #19 testing::UnitTest::Run() external/gtest/googletest/src/gtest.cc:5444:10 (csdp_solver_test+0xe45618)
[0:23:00.206]      #20 RUN_ALL_TESTS() external/gtest/googletest/include/gtest/gtest.h:2293:73 (csdp_solver_test+0xe2717b)
[0:23:00.206]      #21 main common/test_utilities/drake_cc_googletest_main.cc:28:10 (csdp_solver_test+0xe26feb)
[0:23:00.206]  
[0:23:00.206]    Previous read of size 8 at 0x7bb400003458 by thread T1:
[0:23:00.206]      #0 memcpy <null> (csdp_solver_test+0x4a660e)
[0:23:00.206]      #1 dgemm_oncopy_SKYLAKEX <null> (libopenblas.so.0+0x19fc8d3)
[0:23:00.206]  
[0:23:00.206]    Location is heap block of size 35912 at 0x7bb400000000 allocated by main thread:
[0:23:00.206]      #0 malloc <null> (csdp_solver_test+0x49ad1b)
[0:23:00.206]      #1 easy_sdp external/csdp/lib/easysdp.c:274:6 (csdp_solver_test+0x72baaf)
[0:23:00.206]      #2 drake::solvers::internal::csdp::cpp_easy_sdp(char const*, int, int, drake::solvers::internal::csdp::blockmatrix, double*, drake::solvers::internal::csdp::constraintmatrix*, double, drake::solvers::internal::csdp::blockmatrix*, double**, drake::solvers::internal::csdp::blockmatrix*, double*, double*) solvers/csdp_cpp_wrapper.cc:32:10 (csdp_solver_test+0x72b24f)
[0:23:00.206]      #3 drake::solvers::internal::(anonymous namespace)::SolveProgramThroughLorentzConeSlackApproach(drake::solvers::MathematicalProgram const&, drake::solvers::internal::SdpaFreeFormat const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, drake::solvers::MathematicalProgramResult*) solvers/csdp_solver.cc:371:19 (csdp_solver_test+0x718122)
[0:23:00.206]      #4 drake::solvers::CsdpSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const solvers/csdp_solver.cc:499:9 (csdp_solver_test+0x716677)
[0:23:00.206]      #5 drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const solvers/solver_base.cc:82:5 (csdp_solver_test+0xa3ff39)
[0:23:00.206]      #6 drake::solvers::test::RunSolver(drake::solvers::MathematicalProgram const&, drake::solvers::SolverInterface const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&) solvers/test/mathematical_program_test_util.cc:18:10 (csdp_solver_test+0xb1df35)
[0:23:00.206]      #7 drake::solvers::test::TestFindSpringEquilibrium::SolveAndCheckSolution(drake::solvers::SolverInterface const&, std::optional<drake::solvers::SolverOptions> const&, double) solvers/test/second_order_cone_program_examples.cc:359:7 (csdp_solver_test+0xa48525)
[0:23:00.206]      #8 drake::solvers::test::TestFindSpringEquilibrium_TestSOCP_Test::TestBody() solvers/test/csdp_solver_test.cc:309:7 (csdp_solver_test+0x50db3c)
[0:23:00.206]      #9 void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) external/gtest/googletest/src/gtest.cc:2599:10 (csdp_solver_test+0xe63f37)
[0:23:00.206]      #10 void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) external/gtest/googletest/src/gtest.cc:2635:14 (csdp_solver_test+0xe4d9e2)
[0:23:00.206]      #11 testing::Test::Run() external/gtest/googletest/src/gtest.cc:2674:5 (csdp_solver_test+0xe38890)
[0:23:00.206]      #12 testing::TestInfo::Run() external/gtest/googletest/src/gtest.cc:2853:11 (csdp_solver_test+0xe3918e)
[0:23:00.206]      #13 testing::TestSuite::Run() external/gtest/googletest/src/gtest.cc:3012:30 (csdp_solver_test+0xe399b7)
[0:23:00.206]      #14 testing::internal::UnitTestImpl::RunAllTests() external/gtest/googletest/src/gtest.cc:5870:44 (csdp_solver_test+0xe45c47)
[... cut ...]
[0:23:00.206]      #17 testing::UnitTest::Run() external/gtest/googletest/src/gtest.cc:5444:10 (csdp_solver_test+0xe45618)
[0:23:00.206]      #18 RUN_ALL_TESTS() external/gtest/googletest/include/gtest/gtest.h:2293:73 (csdp_solver_test+0xe2717b)
[0:23:00.206]      #19 main common/test_utilities/drake_cc_googletest_main.cc:28:10 (csdp_solver_test+0xe26feb)
[0:23:00.206]  
[0:23:00.206]    Thread T1 (tid=22, running) created by main thread at:
[0:23:00.206]      #0 pthread_create <null> (csdp_solver_test+0x49c54b)
[0:23:00.206]      #1 blas_thread_init <null> (libopenblas.so.0+0x371c78)
```

jwnimmer-tri, 2022-10-13:
I don't understand this report.

The "Location is heap block" makes sense -- that's where we allocate the `O` matrix.

The "Write of size 8" makes sense -- that's where we fill `O` with zeros.  The address we're writing to is in bounds -- it's only about 1/3 of the way through `O`'s block of memory.

The "Previous read of size 8" from another thread doesn't make any sense.  From the moment where we allocate `O` until the "Write of size 8", nothing in the source code touches that pointer.  We only ever pass it from `easy_sdp` to `sdp` to `op_o` and and `op_o` is the first place that accesses it in any way.

Given that escape analysis, I have no idea how a thread would obtain that pointer, nevermind start reading from it.  All I can guess is that the thread is just -- reading from some random address that it does not own, or maybe doesn't own anymore?  Or maybe TSan got confused about memory that was free'd and then re-malloc'd?

jwnimmer-tri, 2022-10-13:
The problem still occurs even if _only_ the one test case in question (`CsdpTest/TestFindSpringEquilibrium.TestSOCP/0`) is run by itself using `--gtest_filter`.  And in particular, only the `kLorentzConeSlack` setting for the `RemoveFreeVariableMethod` is failing.  The other two settings pass, no matter which order the three are run in.  Just `kLorentzConeSlack` by itself in the loop is enough to trigger the report.

Possibly worth noting is that on Jammy our default BLAS appears to be `libopenblas0-pthread`.

jwnimmer-tri, 2022-10-13:
With OpenBLAS [debugging symbols](https://wiki.ubuntu.com/Debug%20Symbol%20Packages):

```
  Read of size 8 at 0x7bb400008a30 by thread T26:
    #0 memcpy <null> (csdp_solver_test+0x4a1dfe)
    #1 dgemm_oncopy_HASWELL 0-pthread/kernel/../kernel/x86_64/dgemm_ncopy_8_skylakex.c:410:18 (libopenblas.so.0+0x176517b)

  Thread T26 (tid=883620, running) created by main thread at:
    #0 pthread_create <null> (csdp_solver_test+0x497d3b)
    #1 blas_thread_init 0-pthread/driver/others/./0-pthread/driver/others/blas_server.c:667:11 (libopenblas.so.0+0x371c78)
    #2 blas_thread_init 0-pthread/driver/others/./0-pthread/driver/others/blas_server.c:628:5 (libopenblas.so.0+0x371c78)
```

https://github.com/xianyi/OpenBLAS/blob/v0.3.20/kernel/x86_64/dgemm_ncopy_8_skylakex.c#L410

https://github.com/xianyi/OpenBLAS/blob/v0.3.20/driver/others/blas_server.c#L667

I wonder how to get a longer backtrace for the Read.

jwnimmer-tri, 2022-10-13:
Force-downgrading the OpenBLAS CPU detection using `env OPENBLAS_CORETYPE=CORE2` avoids the TSan warning, even though the library still pre-launches dozens of helper threads on library start-up.  It's apparently only the OpenBLAS drivers for Skylake, Haswell, etc. that seem to cause a problem by fighting with TSan somehow.

jwnimmer-tri, 2023-01-06:
Mostly likely fixed by https://github.com/RobotLocomotion/drake/pull/18540.

closed 2023-01-06 by jwnimmer-tri