RobotLocomotion/drake #17571: Improve documentation for setting center of mass of a body

allenzren, 2022-07-13:
In the Body/Rigidbody class, users can call `SetCenterOfMassInBodyFrame` to move the center of mass of a body in the body frame. However, this does not shift the spatial inertia of the body to the new center of mass. In contrast, `SetMass` does change the spatial inertia.

Users may assume changing the center of mass automatically shifts the inertia, which is not true. The new spatial inertia becomes physically invalid.

We may add such to the documentation of `SetCenterOfMassInBodyFrame`: "Note that this does not shift the spatial inertia of the body to the new center of mass. Please see `Shift' and `ShiftInPlace' in SpatialInertia class."

allenzren, 2022-07-13:
As a side note, I figured out the issue when the physically invalid spatial inertia caused SAP solver to fail:
`Failure at multibody/contact_solvers/sap/sap_model.cc:391 in CalcDelassusDiagonalApproximation(): condition 'A_ldlt[c].eigen_linear_solver().isPositive()' failed.` @amcastro-tri  TAMSI did not throw an error.

sherm1, 2022-07-14:
I don't think that's a bug, Alejandro, except maybe that we provided "Set" sugar methods when we should have stuck with the minimal SetSpatialInertia API. If someone sets mass properties by calling SetMass, SetInertia, SetCOM in that order I think they would not be pleased if the last call changed the the inertia. Should we deprecate these methods?

I could be convinced otherwise but that's my current thought. What do you propose?

amcastro-tri, 2022-07-14:
SetCOM should also shift the COM leading to a valid model. Maybe the name is misleading. We should not have SetCOM but instead ShiftCOM?
I don't think a user would ever like to call SetCOM and end up with an invalid model?

mitiguy, 2022-11-03:
PR https://github.com/RobotLocomotion/drake/pull/18251 is part of resolution towards issue #17571.

mitiguy, 2023-01-31:
PR https://github.com/RobotLocomotion/drake/pull/18582 resolves https://github.com/RobotLocomotion/drake/issues/17571.

closed 2023-01-31 by mitiguy