RobotLocomotion/drake #17630: Meshcat pointcloud isn't visible in 2D render mode (perspective projection)

RussTedrake, 2022-07-26:
### What happened?

Running the following code:
```
  geometry::Meshcat meshcat;
  meshcat.Set2dRenderMode(math::RigidTransformd(Eigen::Vector3d{0, 0, 1}), -2,
                          2, -2, 2);
  perception::PointCloud test_cloud(4);
  // clang-format off
  test_cloud.mutable_xyzs() << 0, 1, 1, 0,
                          1, 1, 0, 0,
                          0, 0, 0, 0;
  // clang-format on
  meshcat.SetObject("Box corners", test_cloud, 0.1, Rgba(0, 1, 0));
  meshcat.SetLine("Box", test_cloud.xyzs().cast<double>(), 1.0, Rgba(1, 0, 0));
  std::cout << "[Press RETURN to continue]." << std::endl;
  std::cin.ignore(std::numeric_limits<std::streamsize>::max(), '\n');
```
Results in meshcat showing red lines in the intended box location, but no green points.
Commenting out the Set2dRenderMode line results in the visualization appearing as expected.

Frustratingly, the `meshcat_manual_test` does cover this case, and the point cloud looks fine in the perpective proejction.

I'm not sure if Drake is responsible, or meshcat, or three.js, etc.  But I'll open this for awareness and tracking.

AlexandreAmice, 2023-09-06:
Hitting this today. Is there any recommended quick fix?

jwnimmer-tri, 2023-09-06:
I have no experience here, but this comment caught my eye:

> Frustratingly, the `meshcat_manual_test` does cover this case, and the point cloud looks fine in the perspective projection.

The manual test has this:

```
    perception::PointCloud cloud(
        kPoints, perception::pc_flags::kXYZs | perception::pc_flags::kRGBs);
```

I wonder if adding the RGBs per-point would be some kind of a work-around?  Or in general, trying to figure out what the test is doing that works, different from the example up above.