RobotLocomotion/drake #17638: [geometry] MeshFieldLinear has broken public copy semantics

SeanCurtis-TRI, 2022-07-27:
### What happened?

`MeshFieldLinear` makes use of `DRAKE_DEFAULT_COPY_AND_MOVE_AND_ASSIGN`. It also has a `reset_on_copy<Mesh*>` field. This allows the mesh field to be easily, mindlessly copied but left in an invalid state. Any invocation on the mesh that requires reference to the mesh turns into a null pointer dereference.

There is a utility for cloning a `MeshFieldLinear` and accounting for the mesh (`CloneAndSetMesh`). It makes (indirect) use of the copy constructor, so one should be maintained. But it should probably be private.

This came up, for example, in python bindings for the field. Through an oversight, the field was implicitly copied (allowed by the defaulted copy constructor) which led to incorrect bindings.

### Version

1.5

DamrongGuoy, 2022-08-01:
I assume this issue is `priority: medium`.  I'll add it to my Q3 OKRs.

Is this a possible solution?  Change `reset_on_copy<const MeshType*> mesh_;` to `const MeshType* mesh_;` and document that it is a shallow copy.

SeanCurtis-TRI, 2022-08-01:
I'd just make the copy semantics private. If anyone ever wants to clone a field, they should have to call `CloneAndSetMesh()`.

DamrongGuoy, 2022-08-02:
Perhaps I don't understand "make the copy semantics private".  I failed to build my prototype #17670.  Please feel free to comment there. I might miss something in C++.

In the meantime, I also prototype #17671 to remove `reset_on_copy` and wait for CI now.  It built ok on my Puget.

DamrongGuoy, 2022-08-03:
Again thanks for helping me fix #17670. I'll start the review process shortly after cleaning up.

PR #17670 (2022-08-04): [geometry] Fix copy semantics of MeshFieldLinear
Make copy constructors of MeshFieldLinear private.
Use CloneAndSetMesh() instead of the copy constructors.

Fixes #17638

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17670)
closed 2022-08-04 by SeanCurtis-TRI