RobotLocomotion/drake #17649: The BuildInto method of DiagramBuilder does not work with SceneGraph subsystem

huweiATgithub, 2022-07-28:
### What happened?

As I have raised in https://github.com/RobotLocomotion/drake-external-examples/issues/233, I found that the particle example did not draw in the visualizer. The example in the drake-external-examples repo is the same as that in the drake repo: https://github.com/RobotLocomotion/drake/tree/master/examples/particles.

I was noticing that neither the OutputPose callback of the geometry nor the OutputState callback of the particle system (connected to geometry) was being called during the simulation.
Besides, a simple anchored geometry registered to the scene_graph is also not showing.
However, the state of the particle and the calculating of the time derivative is working properly. 
 
After some diggings and comparison with the scene_graph [examples](https://github.com/RobotLocomotion/drake/tree/master/examples/scene_graph), I finally managed to have the particle moved and shown correctly by replacing `builder.BuildInto(this);` in the constructor of UniformlyAcceleratedParticle with `builder.Build();` (and having the simulation inside the constructor).

It seems that the BuildInto method might have some problems in handling stuff with scene_graph.

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

apt install drake

SeanCurtis-TRI, 2022-07-28:
I believe the error can be found here:

https://github.com/RobotLocomotion/drake/blob/master/examples/particles/uniformly_accelerated_particle.cc#L81

If you change the call to `AllocateContext()` to `CreateDefaultContext()` the geometry is then put to work. This is because the geometry data doesn't get copied into the context by allocation.

SeanCurtis-TRI, 2022-07-28:
Specifically, this point to a bug in `SceneGraph` -- it most likely arose when the geometry data stopped being state and became a parameter. The constructed model only gets copied over into the context as a result of calling `SetDefaultParameters()` (which *is* called by `CreateDefaultContext()`). It also needs to be copied when simply allocating the context.

jwnimmer-tri, 2022-07-28:
My vote is that we simply delete the `particle` example.

I **do** think we should have an example which demonstrates dynamics without using a MultibodyPlant, along with illustration geometry -- but I think both `example/acrobot` and `example/pendulum` serve that purpose just fine already.  A particle that accelerates beyond the visualizer within a moment of the start of the simulation isn't terribly useful, to me.

SeanCurtis-TRI, 2022-07-29:
That's fine and good.

However, I introduced an elephant into the room. The claim that `SceneGraph` is broken. However, as a result of #14189 and as noted in the [release notes for v0.24.0](https://github.com/RobotLocomotion/drake/blob/master/doc/_release-notes/v0.24.0.md), it is not/should not be the expectation that `SceneGraph` works when simply *allocating* a context. We should explicitly be calling `CreateDefaultContext()`.

So, in addition to uselessly showing a particle accelerate out of the view, its implementation is an anti-pattern.

I did a quick survey to see if `AllocateContext()` was called anywhere else bad, and it doesn't seem to be. So, nuking this one example seems to be sufficient to eliminate the anti-pattern. I'll rename the issue again and submit a PR to nuke the example in question.

huweiATgithub, 2022-07-29:
I believe the particle example itself is not the problem. It is the simplest example that builds a dynamical system with visualization from scratch.
What is wrong is the way it initializes the context. It seems that changing the way of setting up context will fix it.
If AllocateContext has some problem, I think it should be either hidden from public usage or has the caveats documented properly.

jwnimmer-tri, 2022-07-29:
That's a good point.  The API documentation for `AllocateContext` (on both `SystemBase` and `System`) should remind users that it fills the returned context with NaNs and/or nulls, and to use `CreateDefaultContext` instead if they wanted meaningful values.

The purpose of "Allocate" is for advanced uses where we need to allocate storage ahead of time, for a function that needs an output-pointer to scribble into.  Probably that means "(Advanced) " should precede the `AllocateContext` documentation as well.

jwnimmer-tri, 2022-07-29:
It might be worth considering renaming `AllocateContext` to `AllocateUninitializedContext` for utmost clarity?

jwnimmer-tri, 2023-10-30:
PR #17653 deleted the Drake copy of the broken demo, but not the `drake-external-examples` copy, e.g.:
https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed/src/particles

There are actually a lot of copies:
```
jwnimmer@call-cps:~/jwnimmer-tri/drake-external-examples$ find . -name 'particle.cc' | sort
drake_ament_cmake_installed/src/drake_ament_cmake_installed/src/particle.cc
drake_catkin_installed/src/drake_catkin_installed/src/drake_catkin_installed/particle.cc
drake_cmake_installed_apt/src/particle.cc
drake_cmake_installed/src/particles/particle.cc
```

jwnimmer-tri, 2023-10-30:
Ah, I was slightly confused.  Only the `drake_cmake_installed` copy has the (broken) visualization.  The other copies are just doing an example of LeafSystem dynamics with no visualization.

jwnimmer-tri, 2025-01-17:
https://github.com/RobotLocomotion/drake-external-examples/pull/355 removes the broken example, and #22479 clarifies the docs.

PR #17653 (2022-08-01): [examples] Deleting the particles example
While the particles example illustrates the absolutely simplest possible dynamics system (with a non-zero acceleration), that is its *only* value. Visualizing it shows a sphere instantly racing out of view and the
code manifests an anti-pattern (using `AllocateContext()` instead of `CreateDefaultContext()`).

As there are other examples that introduce bespoke continuous systems (e.g., pendulum), this examples contribution is not justified.

closes #17649

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17653)
PR #22479 (2025-01-17): [systems] Clarify doc that AllocateContext is a trap
Closes #17649.

+@sherm1 for both reviews, please!

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22479)
closed 2025-01-17 by jwnimmer-tri