RobotLocomotion/drake #17661: Add realtime rate chart to meldis

jwnimmer-tri, 2022-08-01:
When a MeshcatVisualizer is connected directly to a Simulator, it can show a nice realtime rate chart (see #16981 for a screencap video).

We should enhance `meldis` to display the same chart, using the LCM message times to populate the rate.

DamrongGuoy, 2022-11-16:
Thank you!  Now I also understand the `73 rtr%` means 73% realtime rate.

This feature would be very helpful when we try to speed up our simulators or evaluate new codes.

jwnimmer-tri, 2023-02-14:
FYI a solution outline (untested):

```patch
--- a/bindings/pydrake/visualization/_meldis.py
+++ b/bindings/pydrake/visualization/_meldis.py
@@ -143,6 +143,13 @@ class _ViewerApplet:
             link_path = f"{self._path}/{robot_num}/{link_name}"
             pose = self._to_pose(message.position[i], message.quaternion[i])
             self._meshcat.SetTransform(path=link_path, X_ParentPath=pose)
+        sim_time = message.timestamp * 1e-6
+        wall_time = time.time()
+        if self._prior_sim_timestamp is not None:
+            sim_delta = sim_time - self._prior_times(0)
+            wall_delta = wall_time - self._prior_times(1)
+            self._meshcat.SetRealtimeRate(sim_delta / wall_delta)
+        self._prior_times = (sim_time, wall_time)
         if self._waiting_for_first_draw_message:
             self._waiting_for_first_draw_message = False
             self._build_links()
```

PR #21534 (2024-07-23): Reduce the noise in meshcat visualizer realtime rate reports
Adds a new API `Meshcat` related to realtime rate. Rather than having clients compute the realtime rate themselves (and immediately broadcasting it to all clients), clients simply report to Meshcat that sim time has advanced.

Meshcat uses the report of time advancement to throttle the report of real time rate (and computes the real time rate itself based on the amount the simulation has advanced since the last broadcast message).

Also changed the display of the realtime rate -- rather than streaming values as fast as it can, it onl [... cut]
closed 2024-07-23 by SeanCurtis-TRI