RobotLocomotion/drake #17662: Visualize inertia ellipsoids

jwnimmer-tri, 2022-08-01:
I'd like to have a checkbox in our visualizer(s) to display the inertia ellipsoids for our MbP's rigid bodies.  This would help users debug their SDFormat models by visualizing the masses & etc to make sure everything's in the right place and of an appropriate scale.

EricCousineau-TRI, 2022-12-01:
Howdy @trowell-tri! Any chance this is in your queue for next couple of weeks?
Excited to try this out to check on models, e.g. on the IIWA, related to [this post in Drake Slack](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1666731346091319) and torque control (landed via #18295, relatedly Anzu issue 9395).

trowell-tri, 2022-12-01:
Yes! I hope to start working on this later next week, and get something usable if not final this month.

EricCousineau-TRI, 2022-12-01:
Awesome, thanks!

RussTedrake, 2022-12-09:
fwiw -- I saw two student bugs just last night related to having center of masses outside of the collision geometry (leading to confusing behavior in sim).  SDF makes it far too easy to make this mistake.  This feature will really help.

RussTedrake, 2022-12-10:
two MORE student projects have had the same issue.  i can't imagine this will land in time to help the projects this semester, but it is badly needed.  

As a one-off, I'm recommending people use something like this to at least visualize the center of mass:
```
    plant_context = plant.GetMyContextFromRoot(context)
    body = plant.GetBodyByName("base_link")
    p_BCom = body.CalcCenterOfMassInBodyFrame(plant_context)
    X_WB = body.EvalPoseInWorld(plant_context)
    p_WCom = X_WB @ p_BCom
    meshcat.SetObject("center_of_mass", Sphere(0.01), Rgba(0, 1, 0, 1))
    meshcat.SetTransform("center_of_mass", RigidTransform(p_WCom))    
```

trowell-tri, 2023-02-20:
Plan of attack:
• Add geometry to scene graph
• Easiest to affix geometry to existing bodies?
• SceneGraph may not support wireframes (but I'm not sure that's the way we want to go anyway)
• Group under a magic name with own alpha slider

jwnimmer-tri, 2023-02-20:
New brainwave from today: by default, we should add the inertia geoms with alpha=0 so they are invisible.  Then, for specific visualizers that opt-in, those visualizer front-ends can offer a control to unhide the geoms.

EricCousineau-TRI, 2023-02-20:
> Or do we just one one box per RigidBody, even if the bodies are welded?

Is it feasible to somehow have both (e.g. two alphas) or make it configurable (e.g. via config structs + operators)?
By default, I assume people care about constituent bodies, not merged, because that's what they're authoring / debugging. However, I can see someone wanting to see what the composite inertia looks like at a given configuration.

jwnimmer-tri, 2023-03-27:
@trowell-tri see https://github.com/jwnimmer-tri/drake/commits/inertia-visualizer

There are three commits.  The first one my the original prototype.  The second one tweaks the original to build vs current master.  The final one is the new logic for `meshcat:accepting` to get the geometry into a different subtree.

jwnimmer-tri, 2023-06-08:
#19210 is in now as a starting point.

Once #19414 is done we can start to attack dealing with the rotations, and then enable the feature by default.

@trowell-tri Any suggestions on what's the next most important step here after that?  Should we try to texture the ellipsoids?  Or ask some users for feedback first?

trowell-tri, 2023-06-08:
I was going to say we should texture the ellipsoids — it makes sense to me that it's better if they're obviously _not_ regular geometry — but asking for feedback first makes sense.

The collision geometry doesn't have a special texture, which makes me wonder how critical texturing the inertia ellipsoids actually is. On the other hand, collision is literally geometry from the model so maybe that shouldn't be compared to the inertia, which is geometry we're _deriving_ from the model.

jwnimmer-tri, 2023-06-21:
For the record, one of the other things that came up during reviews so far are two display options:
- Setting for whether to show a box vs an ellipsoid.
- Setting for whether to show the spatial inertia (incorporating mass) vs the unit inertia (not scaled by mass).

I'm tallying them here for completeness.  Given the rudimentary UI layout options at our disposal, I'm not sure that the Meshcat buttons that would allow toggling those are worth the screen real estate cost.  Possibly we could have specific code in `model_visualizer` to offer those options, but not make it part of the default knobs that come up with every simulation.

closed 2023-07-11 by jwnimmer-tri