RobotLocomotion/drake #17663: Implement parsing of coupler constraints.

amcastro-tri, 2022-08-01:
#17639 introduces coupler constraints between two single-dof joints. This is particularly useful to model things like your parallel jaw gripper where you'd like both finger to move in opposition. 

As pointed out by @EricCousineau-TRI, we might want to have parsing for this in a not very far future.

EricCousineau-TRI, 2022-09-22:
\cc @scpeters @azeey regarding today's discussion about SDFormat's `mimic` feature

RussTedrake, 2023-01-31:
FYI -- I received another feature request for parsing the `mimic` tag today.

RussTedrake, 2023-02-02:
A few relevant links:
- [mimic tag in urdf](http://wiki.ros.org/urdf/XML/joint)
- [mimic in sdformat](https://github.com/gazebosim/sdformat/pull/1166) -- still just a PR
- [AddCouplerConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a71b7c4d7b4af1c97bb6f7dae906d1088) in Drake

amcastro-tri, 2023-12-15:
@RussTedrake implemented URDF parsing in #18728.
@joemasterjohn implemented SDF parsing in #20503.

closed 2023-12-15 by amcastro-tri

scpeters, 2023-12-15:
cross-posting from https://github.com/RobotLocomotion/drake/pull/20503#issuecomment-1858314968

sorry I didn't chime in sooner (I just noticed https://github.com/RobotLocomotion/drake/pull/20503 after it was merged), but you might consider using the `//joint/axis/mimic` / `//joint/axis2/mimic` syntax that has recently been added to SDFormat 1.11 in libsdformat14:

* http://sdformat.org/spec?ver=1.11&elem=joint#axis_mimic
* http://sdformat.org/spec?ver=1.11&elem=joint#axis2_mimic
* http://sdformat.org/tutorials?tut=mimic_proposal&cat=pose_semantics_docs&
* https://github.com/gazebosim/sdformat/pull/1166
* https://github.com/gazebosim/sdformat/pull/1333