RobotLocomotion/drake #17680: [meshcat] Global control of contact quantities

joemasterjohn, 2022-08-03:
Towards #17679

We would like to provide interactive global control of contact quantities in the Meshcat dropdown menus through the use of sliders and buttons. The following options should be supported to match parity with DrakeVisualizer:

### Point Contact
- Vector scaling mode (Fixed, Scaled, Auto-scaled)
- Global scale for fixed length mode
- Minimum magnitude threshold

### Hydroelastic Contact
- Min/Max pressure values
- Turn on/off wireframe (globally or for each surface individually)
- Turn on/off force and moment vectors
- Turn on/off traction vectors
- Turn on/off slip velocity vectors
- Vector scaling mode (Fixed, Scaled, Auto-scaled)
- Global scale for fixed length mode
- Minimum magnitude threshold

jwnimmer-tri, 2022-08-03:
When I investigated this a few months ago, a survey of users indicated that some of these options ended up being unused / not relevant.  I suggest we drop support for those options, instead of re-implementing them.

I'll see if I can dig up the discussion and retcon a link to it here.

SeanCurtis-TRI, 2022-08-03:
I can easily imagine that some of those features are unused (e.g., visualizing traction). Getting that concrete list of the features with the highest utility would be a big help.

jwnimmer-tri, 2022-08-03:
So far, I've only found https://github.com/RobotLocomotion/drake/issues/15738#issuecomment-1035285409 in my notes.  I think everything else I learned got lost in the mists of time.

At worst, we can just do a new survey.  Anything that nobody mentions we can just omit until someone asks for it again?

SeanCurtis-TRI, 2022-08-03:
Whatever the final feature set looks like, I like the `drake_visualizer` full list in the issue. We can be *explicit* about what we've *decided* to leave out. Absence too easily can be mistaken for oversight.