RobotLocomotion/drake #17682: [meshcat] Use better shading for hydroelastic contact patches

joemasterjohn, 2022-08-03:
Towards #17679

Currently contact surfaces in meshcat are using per-vertex coloring, producing bad visual artifacts:
[image]

In DrakeVisualizer this was fixed by putting the color map in a 1-dimensional texture and passing normalized pressure values as texture coordinates for each vertex, resulting in pleasing per-fragment color shading:
[image]

I see two possible ways to achieve the same per-fragment shading in Meshcat:

1. Update Meshcat to provide a datatype and message packing scheme for the Three.js Texture type. Then pass the pressure values as texture coordinates along with the mesh and have the renderer interpolate the texture coords instead of colors. Currently this would require packaging and sending the same texture map for every contact message, unless there is a way to send one to meshcat, store the uuid and reuse it for other meshes.

2. Pass a custom vertex/fragment shader to the Three.js material where the "pressure" attribute is a varying quantity (i.e. interpolated and passed from vertex shader to fragment shader), and the fragment shader does the color mapping. The shader code would never change and would also (currently) have to be passed with every material object for contacts that we send to Meshcat.

SeanCurtis-TRI, 2022-08-03:
It's worth noting that normalizing the per-vertex data will always lead to the same artifacts. The pressure needs to be interpolated across the triangle and the normalized for each fragment -- that's the only away to eliminate this particular rendering artifact.

A new attribute isn't strictly required (drake_visualizer packs the absolute pressure into the texture coordinate), but a custom fragment shader is required.

SeanCurtis-TRI, 2022-08-03:
Another thought: one thing that is missing is any ability to interpret the colors. `drake_visualizer` has the absolute minimum mechanism for doing so, but is still more than here. We might consider providing some mechanism for understanding the scale communicated by the color map.

SeanCurtis-TRI, 2022-08-03:
Another passing thought: If you're already going to do a fragment shader, you *could* forego the texture by hard-coding the color map in the shader. It gets you to the same (short-term) result without having to deal with textures. Obviously, noot as general as, eventually, letting the user define their own color map, but perhaps a shorter distance to better interpolation? However, if the texture is trivial and the shader is the heavy lifting, then I'd go with the texture from day one.

SeanCurtis-TRI, 2023-10-27:
BTW I came across [this `three.js` demo](https://rawcdn.githack.com/mrdoob/three.js/a05dee742ccbbba8d8f8386bca2fc537d3d45727/examples/index.html#webgl_geometry_colors_lookuptable) which is exactly what we want.

joemasterjohn, 2023-10-31:
Thanks for the link @SeanCurtis-TRI. I will try to take a crack at the meshcat PR for this later this week / early next week.