RobotLocomotion/drake #17688: Enhance MeshcatVisualizer to offer collision geometry checkbox

jwnimmer-tri, 2022-08-03:
In #17595, we added collision geometry visualization to meldis.  We should offer the likewise feature to MeshcatVisualizer (including the alpha sliders and enable/disable checkbox control).

Details to follow.

jwnimmer-tri, 2022-08-09:
The current meldis implementation lives at https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/visualization/meldis.py.

The place where we add the new feature to MeshcatVisualizer is https://github.com/RobotLocomotion/drake/blob/master/geometry/meshcat_visualizer.cc.

RussTedrake, 2022-08-29:
This would be great. For completeness: once it is implemented, we will also want to remove the manual implementation of this feature from the "authoring multibody simulation" tutorial.

jwnimmer-tri, 2022-10-25:
(1) Run meldis to see how this feature works.  There are tickyboxes for collision vs visual, and sliders for visual alpha and collision alpha.  Do this using a simulation or example or tutorial, _not_ the model_visualizer.

(2) It's okay for the tickybox names for in the scene tree to differ between meldis and this tool.  Neither tree of names is that stellar anyway currently.

(3) TBD

closed 2022-12-14 by trowell-tri