RobotLocomotion/drake #17699: Ubuntu Wheel Staging in Jenkins

jwnimmer-tri, 2022-08-05:
Please add a https://drake-jenkins.csail.mit.edu/view/Staging/ build for the linux wheels as well.  Then we can update the https://drake.mit.edu/release_playbook.html to instruct the release manger to use Staging for all of the wheel builds, instead of compiling locally.

See also #17692.

svenevs, 2022-09-06:
Job is created on Jenkins, waiting to confirm test jobs complete before adding docs PR to the release playbook.  Right now it is split into mac and linux instructions, presumably we can fold this into one section?

jwnimmer-tri, 2022-09-06:
Yeah, whatever will make the most sense to the person stepping through the instructions will be fine.

PR #17843 (2022-09-07): [doc] Use linux wheel staging in release playbook
Closes #17699.

[Linux staging job results here.](https://drake-jenkins.csail.mit.edu/view/Wheel/job/linux-focal-unprovisioned-gcc-wheel-staging-snopt-mosek-release/1/consoleFull)

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17843)
closed 2022-09-07 by jwnimmer-tri