RobotLocomotion/drake #17715: Drake-specific handling of std::monostate no longer necessary with latest pybind

SeanCurtis-TRI, 2022-08-11:
### What happened?

In #15925, Drake introduced its own bespoke binding of std::monostate. It was subsequently moved to its own file in #16407. See [monostate_pybind.h](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/common/monostate_pybind.h).

It seems, that pybind itself has caught up. pybind/pybind11#3818 seems to have resolved the initial issue pybind/pybind11#3019 (although it isn't marked as such yet). This patch went in in March 2022. We last updated our fork December 2021.

I modified our bespoke version to match what is in pybind and confirmed that our tests pass. 

So, updating our pybind11 fork should allow us (even *force* us) to remove this band aid.

EricCousineau-TRI, 2022-08-11:
Thanks for the heads up! Will prolly take a while before pybind11 fork update doesn't get clouded out by other priorities.

jwnimmer-tri, 2024-03-13:
Fixed in #20362.

closed 2024-03-13 by jwnimmer-tri