RobotLocomotion/drake #17720: Cloning a SceneGraph Context can potentially bake a QueryObject in an thread-unsafe manner

SeanCurtis-TRI, 2022-08-11:
### What happened?

In `SceneGraph`'s default context, there is a cache entry associated with the `QueryObject` output port. The model and the initial value in that cache entry is a *default*-constructed `QueryObject`. Cloning it is dirt cheap. However, once any method on the `QueryObject` has been evaluated, it is no longer default, but it is considered "live".

To adhere to `AbstractValue`-compatible semantics, one can create a local copy of the `QueryObject` that is "baked". The a copy of all of `SceneGraph`'s state is copied and stored in the `QueryObject`. It no longer depends on the `Context` from which it came and will always report fixed answers based on the state.

The problem is that when we `Clone()` the `Context`, ultimately, we `Clone()` the `QueryObject` in the corresponding `CacheEntry`. So, the cloned `Context` has a *baked* `QueryObject` stored in its `CacheEntry`. We've spent time duplicating the old context's values and have stored them deep inside the new context.

Apparently, this is only a transitory issue, because whenever the `SceneGraph` evaluates its `QueryObject` output port, it sets it to be live w.r.t. its `Context` and the `SceneGraph`. So, it might appear that it's largely harmless -- there is wasted work, but the copies of old data will never see the light of day.

But there is a problem.

If we take such a `Context` (with a live `QueryObject` in the output port's cache entry, and we attempt to clone that context in parallel across multiple threads, then multiple threads will all attempt to update the same cache entry and things go BOOM!

Possible solutions:

- When `a `Context` gets cloned, the `QueryObject` stored in its cache entry should always simply be a default query object. This eliminates the wasted evaluation and copying and guarantees that copying the cached query object is 100% threadsafe.
- "Baking" a query object should get a mutex such that for a single context, only one baking operation can take place at a time (no matter how many references to the `QueryObject` are referenced.
- Some third option?

SeanCurtis-TRI, 2022-08-11:
Note: this bug was exposed in CI:

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-debug/2091/

Specifically, `//multibody/plant:multibody_plant_reaction_forces_test` is the code that clones a context with a live `QueryObject` in multiple threads simultaneously leading to the error:

```
CacheEntryValue(::_::scene_graph:Cache guard for configuration updates)::GetMutableAbstractValueOrThrow(): the current value is already up to date.
```

SeanCurtis-TRI, 2022-08-11:
cc @jwnimmer-tri I'd appreciate your thoughts on this.

rpoyner-tri, 2026-11-11:
Reassigning to @SeanCurtis-TRI (cc @xuchen-han). This is not fixed: #17721 only added a test workaround (the `TODO(#17720)` is still in `multibody_plant_reaction_forces_test.cc`), and `QueryObject`'s copy still bakes state in `query_object.cc`. I'm not working on it. Please unassign yourself or push back if that's wrong.