RobotLocomotion/drake #17732: fmt: brew has upgraded to `fmt@9.0`; needs accommodation in Drake?

EricCousineau-TRI, 2022-08-17:
### What happened?

Per diagnosis from @svenevs on Slack: https://drakedevelopers.slack.com/archives/C270MN28G/p1660741468140729

> From what I can tell the mac job failures stem from [brew recently updating to fmt@9.0](https://github.com/Homebrew/homebrew-core/commits/master/Formula/fmt.rb) (from 8.x), but did not investigate closely.  Either code changes in drake or workspace changes for fmt will be needed it seems.
> [...]
> it's not clear to me which direction is right, but I kind of think it would be less resistance to fix drake to work with fmt 9 and vendor it as a workspace dependency -- it'll be challenging to use brew to install an older version of something

### What operating system are you using?

macOS 11 (Big Sur)

### Relevant log output

```shell
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-big-sur-unprovisioned-clang-bazel-nightly-release/555/consoleText

[2022-08-17T07:41:03.865Z] [31m[1mERROR: [0m/Users/bigsur/workspace/mac-big-sur-unprovisioned-clang-bazel-nightly-release/_bazel_bigsur/bec0ae6dc20cf0d6dbe6178cb16ef78f/external/dreal/dreal/symbolic/BUILD.bazel:29:17: Compiling dreal/symbolic/symbolic.cc failed: (Aborted): wrapped_clang_pp failed: error executing command external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 -DNDEBUG ... (remaining 75 arguments skipped)
[2022-08-17T07:41:03.865Z] 
[2022-08-17T07:41:03.865Z] Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[2022-08-17T07:41:03.865Z] In file included from external/dreal/dreal/symbolic/symbolic.cc:8:
[2022-08-17T07:41:03.865Z] In file included from external/dreal/dreal/util/exception.h:3:
[2022-08-17T07:41:03.865Z] In file included from external/fmt/include/_usr_local_Cellar_fmt_9.0.0_include/fmt/format.h:48:
[2022-08-17T07:41:03.865Z] [1mexternal/fmt/include/_usr_local_Cellar_fmt_9.0.0_include/fmt/core.h:1732:3: [0m[0;1;31merror: [0m[1mstatic_assert failed due to requirement 'formattable' "Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt"[0m
[2022-08-17T07:41:03.865Z]   static_assert(
[2022-08-17T07:41:03.865Z] [0;1;32m  ^
[2022-08-17T07:41:03.865Z] [0m[1mexternal/fmt/include/_usr_local_Cellar_fmt_9.0.0_include/fmt/core.h:1753:10: [0m[0;1;30mnote: [0min instantiation of function template specialization 'fmt::v9::detail::make_value<fmt::v9::basic_format_context<fmt::v9::appender, char>, const dreal::drake::symbolic::Formula &>' requested here[0m
[2022-08-17T07:41:03.865Z]   return make_value<Context>(val);
```

jwnimmer-tri, 2022-08-17:
In our macOS binary release (tgz), we link fmt dynamically, so bleeding-edge Homebrew is not compatible with our most recent stable release [v1.6.0](https://github.com/RobotLocomotion/drake/releases/tag/v1.6.0).

```
macsim:lib jwnimmer$ otool -l libdrake.so | grep fmt
         name /usr/local/opt/fmt/lib/libfmt.8.dylib (offset 24)
```

Users should avoid upgrading homebrew if they intend to use Drake v1.6.0.

Once we fix Drake to be compatible with fmt 9, we should make another stable release.

svenevs, 2022-08-17:
I'm not sure how to proceed on this one, refs https://github.com/fmtlib/fmt/issues/2768#issuecomment-1040594969

1. `bazel build --define FMT_DEPRECATED_OSTREAM=1 //...` should be adding the desired compile time definition, right?  If so, this switch does not work for us -- I assume because fmt was not compiled this way via homebrew.
2. AFAICT the alternative `template <> struct fmt::formatter<SomeClass> : ostream_formatter` is not suitable for all of the usecases.  The one that came up specifically

    https://github.com/RobotLocomotion/drake/blob/a61a1846ff3b0e49e66f3f92beac29121adb32dc/multibody/tree/rotational_inertia.h#L163-L164

    https://github.com/RobotLocomotion/drake/blob/a61a1846ff3b0e49e66f3f92beac29121adb32dc/multibody/tree/rotational_inertia.h#L1070-L1071

    via

    https://github.com/RobotLocomotion/drake/blob/a61a1846ff3b0e49e66f3f92beac29121adb32dc/multibody/tree/spatial_inertia.cc#L51-L52

    It's not obvious how to specialize `fmt::formatter` when I need a `template <typename T>` for `RotationalInertia<T>`?

For some things like `GeometryId id` I can just `id.get_value()`, but from what I'm seeing here the prescribed fix to use deprecated behavior won't help and we're going to have to either change or stop using fmt in some places (e.g., in the above `std::ostringstream` could be used?).

jwnimmer-tri, 2022-08-17:
Bazel's `--define` flag is not preprocessor definitions, it's build variant configs (e.g., MOSEK enabled or not).

See here for fmt preprocessor:

https://github.com/RobotLocomotion/drake/blob/a61a1846ff3b0e49e66f3f92beac29121adb32dc/tools/workspace/fmt/package.BUILD.bazel#L27-L35

jwnimmer-tri, 2022-08-18:
Re-opened to track Wheel build fixups.

svenevs, 2022-08-18:
Nightly production [re-ran successfully](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-big-sur-unprovisioned-clang-wheel-nightly-snopt-mosek-release/63/consoleFull), marking this as resolved again :crossed_fingers:

PR #17739 (2022-08-18): [workspace] Support fmt v9 on macOS
Fixes #17732.

See also `FMT_DEPRECATED_OSTREAM` in the release notes: https://github.com/fmtlib/fmt/releases/tag/9.0.0

Homebrew has updated to fmt 9, but drake's use of `fmt` relies on the deprecated ostream behavior.  Eventually drake will need to stop relying on fmt automatically discovering `std::ostream operator<<` and either specialize `fmt::formatter` to inherit from `ostream_formatter` or for simple things like `GeometryId id` we can use `id.get_value()`.

This should work with current provisioned AND unprovisioned macOS CI, currently provisioned ma [... cut]
closed 2022-08-18 by svenevs