RobotLocomotion/drake #17734: MbP should confirm ownership of elements it operates on

jwnimmer-tri, 2022-08-17:
Consider the following MbP function that accepts a `const Body<T>& body` reference from the user:

```c++
template <typename T>
const std::vector<geometry::GeometryId>&
MultibodyPlant<T>::GetVisualGeometriesForBody(const Body<T>& body) const {
  return visual_geometries_[body.index()];
}
```

If the user made a mistake and the `Body<T>` object was actually an element of some _other_ MbP instance instead of `this` plant, the function would do _something_ but probably not at all what was desired.  (It's not uncommon to create multiple MbP objects during motion planning, in contrast of a simulation where there's often only 1 MbP ever created.)

For all functions that accept a user-provided reference to a multibody element, we should check that the element is actually owned by `this` plant, before we operate on it.

I think we can probably do that by checking `DRAKE_THROW_UNLESS(&body == &get_body(body.index());` in essence, although perhaps we'd like a nicer error message in this case, or perhaps we could think of a templated helper function that could check any element type (body, frame, etc.).

SeanCurtis-TRI, 2022-08-17:
This stems from the `Body` vs `BodyIndex` conversation, yes?

If this same API simply took `BodyIndex` instead of `Body` surely you have the same danger, but without recourse. If you want an *index* to transport across MbP instances, then you are already presupposing that it references *equivalent* bodies.

We have a partition of MbP APIs; some take `MultibodyElement` other take `MultibodyElement::ElementIndexType` -- there is typically no overload bridging the divide. The former APIs check ownership, the latter don't. As such, it's not like the user can say, "I'm unsure, I'd like my work checked" or "I'm an advanced user, I know what I'm doing. Please don't check." The APIs have been divided on what may be (or merely appear to be) an arbitrary dividing line.

jwnimmer-tri, 2022-08-17:
> This stems from the Body vs BodyIndex conversation, yes?

Yes.

The kind of typo this error-check would catch is:

```c++
BodyIndex body_index= ...;
plant_A.DoOperation(..., plant_A.GetBody(body_index), ...);
plant_B.DoOperation(..., plant_A.GetBody(body_index), ...);
```

Note the erroenous "plant_A" in the right half of the second line.

> The APIs have been divided on what may be (or merely appear to be) an arbitrary dividing line.

I'm not sure why that's relevant to this ticket.  If the user _is_ using a MultibodyElement API, shouldn't we check for sanity?

I'm not trying to argue that we should add Element-taking overloads for Index-taking functions as part of this ticket.  I just think the Element-taking overloads that we already have should detect mis-use.

SeanCurtis-TRI, 2022-08-17:
As I read your example, the ambiguity creeps in because my understanding is that an alternative, but acceptable spelling, would be:

```c++
BodyIndex body_index = ...;
plant_A.DoOperation(..., body_index, ...);
plant_B.DoOperation(..., body_index, ...);
```

If that's true, then the "erroneous" example you give feels *equally* correct. Reasoning is as follows:

- First, I'm assuming the signature of `DoOperation()` takes a const reference.
- In your original case, if I had to infer *why* a user would commit such a mistake, I'd imagine it is because the developer is looking to keep `plant_A` and `plant_B` in sync vis a vis the body indexed by `body_index`. The key here is that the mistake is *not* in providing the same `body_index`, but in grabbing the body from the wrong plant.
- The fact that `body_index` apparently *means* the same thing in both plants is a necessary pre-condition for that intent to be satisfied. (Things fall apart quickly if my inferred intent is incorrect.)
- From that perspective, getting a body from `plant_A` that reports the same `body_index` as the body from `plant_B` doesn't preclude `plant_B` from operating on that index.
- Therefore, whether I pass in `body_index` directly or I pass in a body from a different plant that nevertheless reports the same body index, the end result is the same. Therefore it doesn't look like a bug.

All of that assumes that passing a `Body` is merely a transport "convenience" for communicating a body index.

Now, here's my counter-argument to that position.

- Passing a body implies dependency on the *body* (arbitrary properties accessible via that type). The example is only "bug free" if you know that `DoOperation` really only cares about the body's index. Documenting that is cumbersome (when simply changing the parameter to `BodyIndex` announces loudly and clearly what it depends on.
- Not every operation that takes a `Body` would necessarily be safe to consume a `Body` from another plant -- even if the two plants are copies of each other. Documenting that on a function-by-function basis is not a place we would want to be.

So, for those reasons, it might make sense to make a distinction between passing a `BodyIndex` (such that the caller is responsible for knowing it's the index they mean) and passing a `Body` (where the plant throws if it doesn't own the `Body`). But does that imply any kind of convention that a `Body` should never be passed if the implementation only wants its index? (There are a number of interfaces where that exact thing happens; `Body` is just transport for its index.)

That's all a long-winded way to say, the significance of the problem isn't clear to me.

calderpg-tri, 2022-08-18:
> - First, I'm assuming the signature of DoOperation() takes a const reference.
> - In your original case, if I had to infer why a user would commit such a mistake, I'd imagine it is because the [...]

Yes, the case where this shows up in our code is in the anzu/multiarm interface, where we have multiple plants that have been built from the same directives and thus have the same indices, and then those indices are are used to identify bodies and frames across plants. I would expect non-TRI users have similar uses.

> - From that perspective, getting a body from plant_A that reports the same body_index as the body from plant_B doesn't preclude plant_B from operating on that index.
> - Therefore, whether I pass in [...]

Yes, this is why the typo'd versions works in most cases, the `Body` is only being used to specify its `BodyIndex` (and sometimes name for logging) and these are the same between plants.

> - Passing a body implies dependency on the body (arbitrary properties accessible via that type). The example is only "bug free" if you know that DoOperation really only cares about the body's index. [...]

This is exactly the problem, whether or not the implementation of these operations uses anything more than `body.index()` is not something we specify (and shouldn't).

> But does that imply any kind of convention that a Body should never be passed if the implementation only wants its index? (There are a number of interfaces where that exact thing happens; Body is just transport for its index.)

I would agree with such a convention in principle, but I could easily believe that the current structure of `Body`->`BodyIndex` in many of these operations is purely an implementation detail and encoding that into the API doesn't seem like an improvement.

SeanCurtis-TRI, 2022-08-18:
As a random, follow-up thought -- something to provoke discussion -- it almost makes me propose the idea that MbP APIs should never take bodies, but just body indices (MultibodyElementIndex). That is the best way to guarantee that it is always acting on the body (MultibodyElement) that it owns.

I'd carve out an exception to the (body name, model instance index) pair. 

But the principle is don't pass pointers to owned things, pass agnostic representations.

Discuss.

jwnimmer-tri, 2022-08-18:
I think that would be incredibly awkward.  There's a reason we pass the Element reference, because that's what people often have at hand already, e.g., GetBodyByName returns a Body, not a BodyIndex.  For the typical case of just having 1 MbP, using Index instead of Element means we would end up with `.index()` clutter all over the code.

It actually doesn't help the overall problem space anyway.  The specific example discussed above with two _identical_ MbP objects (such that a BodyIndex in interchangeable across both plants) is atypical.  Typically, when a user has two plants, they will differ, and asking them to use an Index actually _increases_ the chance of a mistake going undetected.

Or maybe I'm wrong.  If you want to make an argument for changing the API, you can work up some specific examples that demonstrate the improvement.  Speculating in the absence of specifics examples with reference to existing code is not terribly useful.

But in any case, please do it in a new issue.  This issue is supposed to be simple -- when an function is given bad data, it should fail-fast, especially when doing so is cheap.