RobotLocomotion/drake #17735: Meldis does not update the collision geometry from ManipulationStation

JoseBarreiros-TRI, 2022-08-17:
### What happened?

Ran [joint_teleop](https://github.com/RobotLocomotion/drake/blob/master/examples/manipulation_station/joint_teleop.py).py from manipulation station and Meldis for visualization. Meldis does not seem to update the collision geometry. Perhaps it's a problem with the piping of the visualizer.
 
[image]

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2022-08-17:
Please provide the exact commands and command-line arguments to reproduce the problem.  The manipulation_station programs have a lot of command-line options that substantially change the programs' behavior.

JoseBarreiros-TRI, 2022-08-17:
`bazel run //examples/manipulation_station:joint_teleop`

JoseBarreiros-TRI, 2022-08-17:
bazel run //examples/manipulation_station:joint_teleop

SeanCurtis-TRI, 2022-08-17:
It feels like this must be on me. Let me know if you'd like me to resolve this.

JoseBarreiros-TRI, 2022-08-17:
I ran a modified [version](https://github.com/JoseBarreiros-TRI/drake/tree/meldis_test_manip_station) of ManipulationStation that has a kBoxCollision geometry for the iiwa by default. 
`bazel run //tools:meldis -- -w`
`bazel run //examples/manipulation_station:joint_teleop `
sorry for not adding these details.

jwnimmer-tri, 2022-08-17:
I don't see any collision geometry when I run those commands (with that patch), not would I expect to.  To see collision geometry alongside illustration geometry, some code would need to call either `ApplyVisualizationConfig()` or `AddDefaultVisualization()`, and neither of those calls appear within the `manipulation_station` code anywhere.

Is it possible that you had a long-running meldis visualizer, which you used at one point to visualize the iiwa with collisions via some other commands, and then never restarted meldis prior to running those two commands?

JoseBarreiros-TRI, 2022-08-17:
Perhaps that was the cause of phantom collision geometry. 
I included AddVisualizationConfig() in [joint_teleop](https://github.com/JoseBarreiros-TRI/drake/blob/meldis_test_manip_station/examples/manipulation_station/joint_teleop.py) and it throws this error: `    ApplyVisualizationConfig(
RuntimeError: DiagramBuilder: Cannot operate on ports of System scene_graph until it has been registered using AddSystem`. What would be the right way to use meldis to visualize collision geometries in the manipulation station?

jwnimmer-tri, 2022-08-22:
I'll look into adding collision visualization to that example, but the manipulation_station folder does things in an unusually boutique way compared to everything else in Drake, so it might take me a few days to find the time to dig in and re-learn how it all works.

jwnimmer-tri, 2022-11-09:
Perhaps unsurprisingly, I never got around to updating `manipulation_station`.  I'm putting this on the back burner for the moment.

jwnimmer-tri, 2024-07-09:
... and realistically, nobody is going to work on improving `drake/examples/manipulation_station` anymore.

closed 2024-07-09 by jwnimmer-tri