RobotLocomotion/drake #17736: Jammy support in Drake APT site

jwnimmer-tri, 2022-08-17:
(Details to follow.)

jwnimmer-tri, 2022-09-29:
The apt-upload script has been updated for Jammy and source-controlled in S3.

Next step is to run it on the v1.8.0 tarball and push that to apt (assuming it works).

Then also version control the script it in `drake.git`.  For the secret handling tricks, see `relnotes.py` or `new_release.py` each handling of dotfiles.

BetsyMcPhail, 2022-10-13:
When trying this script on the 1.9.0 release, it failed here: 

https://github.com/RobotLocomotion/drake/blob/df83323bf129cca539174c64505804e0793fe2e5/tools/release_engineering/repack_deb.py#L49

I'm running the release script on a Focal machine _but_ the repack_deb.py script requires a Jammy distribution. @svenevs any insight here? Is this a real requirement?

Also, we should update the comments in that script as they only mention focal.

BetsyMcPhail, 2022-10-13:
Per f2f, I'm going to try adding an optional `codename` argument to the repack_debug.py script. The default (used by CI) will remain the same as it is now.

BetsyMcPhail, 2022-10-13:
Turns out it's not that simple, see #18102.

BetsyMcPhail, 2022-10-13:
There's basically 3 parts of the docker / apt / s3 release process:

1. Run the `push_release` script for docker and tar
2. Create focal and jammy apt packages
3. Run the `push_release` script to sign and push apt releases

Prior to adding Jammy, all 3 steps could be run on any Focal machine. Now, this process will require access to Jammy and Focal. 

In step 2, the Jammy apt package needs to be created on Jammy and the Focal apt package needs to be created on Focal. After they are created they are pushed to S3. In step 3, the `push_release` script gets both of the .deb files from S3. This design is leftover from the old way of creating apt packages. Steps 1 and 2 can be completed anywhere (even on a Mac!).

Although it added another step, this process worked fine for the 1.9.0 release (#18064).

I'm closing the PR that modifies `repack_deb.py`. Working towards Staging builds seems like a better long-term plan.

svenevs, 2022-10-18:
4 staging jobs were added and are going through their first test runs currently.  I'm going to look at the mac artifacts but I don't think we gain anything from the mac `snopt-mosek-packaging` staging jobs (if so will delete).  The linux jobs should provide us good `.tar.gz` and `.deb` artifacts that a release manager can use rather than repacking themselves.

- https://drake-jenkins.csail.mit.edu/view/Staging/job/linux-focal-unprovisioned-gcc-bazel-staging-snopt-mosek-packaging/1/
- https://drake-jenkins.csail.mit.edu/view/Staging/job/linux-jammy-unprovisioned-gcc-bazel-staging-snopt-mosek-packaging/1/
- https://drake-jenkins.csail.mit.edu/view/Staging/job/mac-big-sur-unprovisioned-clang-bazel-staging-snopt-mosek-packaging/lastBuild/
- https://drake-jenkins.csail.mit.edu/view/Staging/job/mac-m1-monterey-unprovisioned-clang-bazel-staging-snopt-mosek-packaging/1/

jwnimmer-tri, 2022-10-18:
Down the road, I could foresee us switching the stable tgz release process to use Staging instead of Nightly, so that we can get a version number into the output instead of YYYYMMDD.

svenevs, 2022-10-19:
The above jobs were just smoke tests to make sure they ran / upload to the right place.  The primary goal is to make it so the release manager doesn't need to e.g., go find a jammy machine just to run `repack_deb`.  I believe the bare minimum requirement to solve this is [update `drake-ci` to request the `--version` for `repack_deb`](https://github.com/RobotLocomotion/drake-ci/blob/f56e639a9e62e788bfa1f33dab394fcefb1d04f0/driver/configurations/bazel/step-create-debian-archive.cmake#L40) (when `staging` is detected).

However, the workflow is a little awkward -- to get the `.deb` you run a staging job that creates a `.tar.gz` only to repack and then it's just ignored.  If we additionally fix the [naming of `.tar.gz` to match nightly when staging is detected](https://github.com/RobotLocomotion/drake-ci/blob/f56e639a9e62e788bfa1f33dab394fcefb1d04f0/driver/configurations/bazel/step-create-package-archive.cmake#L65), the `.tar.gz` produced by these staging jobs would match the names currently used and the release manager just runs the staging packaging jobs (at least for linux, mac value still not apparent) -- as long as they're using the same commit, we don't need to worry about the `.tar.gz` actually getting tested?

> so that we can get a version number into the output instead of YYYYMMDD

Yeah this was what popped into my mind, if you're talking about `share/doc/drake/VERSION.TXT` then instead of `20221018213443 eb5228a98c8a218d6d4d309f4c6b8a37eabfcd15` we would be able to use staging to create `${DRAKE_VERSION} ${sha}` easily, and the mac jobs become justified.  If we do that, instead of naming the `.tar.gz` as `drake-${date}-${codename}.tar.gz` we could also start producing `drake-${version}-${codename}.tar.gz` (rather than matching the nightly naming scheme).  The `repack_deb` script may need to get some simple asserts for not finding `yyyymmdd`.

While this would only affect new release artifacts (not nightlies), it would break anybody who is parsing `VERSION.TXT` expecting a date there.  Not really sure how much of a problem that would or would not be, I think most users would be happy to have an actual numerical version somewhere (even if it's a new file like `RELEASE_VERSION.TXT` with just `x.y.z` in it)?

Either way there's some time between now and the next release to ponder before making decisions on any of this, wanted to get your thoughts before going any further on this vector.

closed 2022-10-20 by svenevs