RobotLocomotion/drake #17738: Review documentation of Optitrack systems and provide examples

JoseBarreiros-TRI, 2022-08-17:
This is related to the [Optitrack Receiver ](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1sensors_1_1_optitrack_receiver.html)and [OptitrackLcmFrameSender](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1sensors_1_1_optitrack_lcm_frame_sender.html) systems. I believe the documentation is a bit cryptic. An example of how to use these systems with a MbP and LCMPublisher system and Python bindings would greatly enhance usability. 
The example could be based on [this](https://github.com/JoseBarreiros-TRI/drake/blob/rl_cito_station_rb/examples/rl_cito_station/rl_cito_station_hardware_interface.cc).

@sammy-tri

sherm1, 2022-08-18:
Assigning to Sam for disposition.

jwnimmer-tri, 2022-08-22:
One resolution here would be to decide that the optitrack-lcm support in Drake has become vestigial and isn't really appropriate for our examples anymore, and therefore to deprecate and remove the code (moving it to Anzu to continue to support the rare times that TRI still uses optitrack).

jwnimmer-tri, 2023-06-14:
In that light, possibly #19600 will make this moot.

jwnimmer-tri, 2023-09-10:
Shall we close this?  There will be no more optitrack code in Drake in <1 month.

JoseBarreiros-TRI, 2023-09-11:
Yes, that seems reasonable.
thanks.

closed 2023-09-11 by jwnimmer-tri