RobotLocomotion/drake #17742: Undo the need for FMT_DEPRECATED_OSTREAM

jwnimmer-tri, 2022-08-18:
To solve #17732 we added `-DFMT_DEPRECATED_OSTREAM` in #17739, but this is not an ideal work-around -- and in any case, will no longer work as of fmt 10.  We will need to rework Drake to concurrently support fmt ~6.1 (what Ubuntu 20 ships) and fmt 9.

jwnimmer-tri, 2023-05-22:
For the record, the issue title doesn't quite capture what remains (though the checklist does).

The remaining work here is to (1) offer better formatters, and (2) deprecate more `operator<<` stuff.  In short, just port Drake to modern fmt best practices.

j4yyousi, 2026-01-02:
Hi Jeremy, I will submit a separate PR for each `Class::operator<<` deprecation for easier review.

closed 2026-03-23 by jwnimmer-tri