RobotLocomotion/drake #17756: Broken Link in Gallery

EvelynDixon-TRI, 2022-08-19:
### What happened?

The reference link to the source code for the trajectory optimization used in the implementation of DIRCON is out of date. 
This is referenced at:  https://drake.mit.edu/gallery.html under the Trajectory Optimization section.

The correct link should be:
https://github.com/DAIRLab/dairlib/tree/master/systems/trajectory_optimization/dircon

jwnimmer-tri, 2022-08-22:
To edit the gallery:
- Browse to the [gallery.md source code](https://github.com/RobotLocomotion/drake/blob/master/doc/_pages/gallery.md).
- Click the "Edit" pencil on the right.
- Make the changes (and use the "Preview" button if desired).
- Choose "Create a new branch" and give it a name:
   [image]
- Click "Propose Changes" to open a pull request.
- The on-call platform reviewer will notice the PR and review it.

EvelynDixon-TRI, 2022-08-23:
Jeremy,
Should I take this issue and follow your recommendations?
Evelyn

-- 
[image: Toyota Research Institute Logo] <https://www.tri.global/>
Evelyn R Dixon (she/her)
エブリン ディクスオン
Product and Program Management
M +1 408 859 1669 <+14088591669>
TOYOTA RESEARCH INSTITUTE <http://www.tri.global/>
Los Altos <https://goo.gl/maps/3MF8fCYh8vV5YMHE6>  / Cambridge
<https://goo.gl/maps/9QWtf6NV6oNrdJih8>

jwnimmer-tri, 2022-08-23:
That's fine by me. I don't plan to work on it.

closed 2023-01-18 by sherm1