RobotLocomotion/drake #17765: Add SDFormat and URDF parsing for screw joint

jwnimmer-tri, 2022-08-23:
See https://github.com/RobotLocomotion/drake/issues/12404#issuecomment-1111679692

jwnimmer-tri, 2022-08-23:
+@scpeters is on deck for this one -@rpoyner-tri.

scpeters, 2022-09-01:
I've started working on this and quickly realized that the `//joint/thread_pitch` parameter is not clearly defined in the SDFormat specification. I've opened https://github.com/gazebosim/sdformat/issues/1125 to address this.

scpeters, 2022-09-02:
I also notice that Drake's screw joint does not accept an axis unit vector parameter; it is defined to rotate and translate [along the Z-axis of the F and M frames](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/screw_joint.h#L20-L25). This raises the question of how to handle the `//joint/axis/xyz` parameter. For initial testing, we can use a model with `//joint/axis/xyz == "0 0 1"`, but for models that specify a different joint axis I can think of two options:

* Rotate the F and M frames so that their Z-axis aligns with the value specified in `//joint/axis/xyz`. This does not require any changes to drake, but it causes F and M frames to deviate from the joint frame defined in SDFormat
* Add support for a joint axis unit vector parameter to `multibody::ScrewJoint`

sherm1, 2022-09-02:
My first thought would be treat this the same way we do a rotational or prismatic joint. Do we rotate the F and M frames for those? That seems reasonable to me. Is it likely that someone would define a joint frame that doesn't line up with the axis?

scpeters, 2022-09-02:
> My first thought would be treat this the same way we do a rotational or prismatic joint. Do we rotate the F and M frames for those?

the [RevoluteJoint](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/revolute_joint.h#L49-L57) for example accepts a `Vector3d` axis parameter, but [ScrewJoint](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/screw_joint.h#L67-L70) does not have a similar axis parameter

scpeters, 2022-09-09:
I just opened a feature request to add a `Vector3` axis parameter to the ScrewJoint constructor in #17882

scpeters, 2022-10-18:
draft pull request in https://github.com/RobotLocomotion/drake/pull/18127 that depends on https://github.com/gazebosim/sdformat/pull/1168

scpeters, 2022-10-27:
pull request for URDF parsing in https://github.com/RobotLocomotion/drake/pull/18191

PR #18191 (2022-11-03): [multibody] Parse screw joints from URDF
Closes #17765.

This adds support for parsing `screw` joints from URDF using a custom `<drake:joint/>` element. The screw thread pitch is specified with a `value` attribute in a custom `<drake:screw_thread_pitch/>` element. To maintain parity with the SDFormat parsing test, a continuous joint is added to the URDF parsing test to match the changes to the SDFormat test from #17956.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18191)
closed 2022-11-03 by EricCousineau-TRI