RobotLocomotion/drake #17784: [meshcat] port "PoseSliders" from underactuated/manipulation course notes

RussTedrake, 2022-08-29:
Earlier this year, @jwnimmer-tri ported the python meshcat [JointSliders](https://github.com/RobotLocomotion/drake/blob/7dd3fe593d8817d83b720214cb4b78f7b8aeaf5e/multibody/meshcat/joint_sliders.h) from my course repos into Drake in order to support the multibody tutorial.  I also have a meshcat [PoseSliders](https://github.com/RussTedrake/manipulation/blob/08d8507821e192f68786f7ecdaf71b00ea3dbdc8/manipulation/meshcat_utils.py#L66) class which is almost identical (but outputs a `RigidTransform` instead of a `VectorXd` of joint positions).  

We already have one version of this implemented in [`//examples/manipulation_station:end_effector_teleop.py`](https://github.com/RobotLocomotion/drake/blob/7dd3fe593d8817d83b720214cb4b78f7b8aeaf5e/examples/manipulation_station/end_effector_teleop_sliders.py#L38).  But it deserves to be a first-class citizen, alongside `JointSliders`.  In the initial PR, it would simply replace the `ManipulationStation` implementation, but down the line it would be useful to have a differential IK tutorial, like the ["intro" notebook](https://deepnote.com/workspace/Manipulation-ac8201a1-470a-4c77-afd0-2cc45bc229ff/project/1-Introduction-6e32ce10-e406-4026-abfb-00be78c478cb/%2Fintro.ipynb) from my manipulation notes.

rpoyner-tri, 2022-08-29:
I'll let @jwnimmer-tri be the task gatekeeper, but this sounds like something that @trowell-tri might be interested in.

jwnimmer-tri, 2023-04-06:
See also #19153.

We'd probably want this to be `pydrake.visualization.PoseSliders` even though `JointSliders` is somewhere else for now.

commit 1eae5139 (2023-06-08): Implement MeshcatPoseSliders (#19513)

This provides a Systems interface for using sliders in meshcat to
create a pose on the output port. This has been used extensively in my
manipulation course notes (e.g. for end-effector teleop in meshcat),
and will soon power an upcoming tutorial in Drake. This is closely
related to the existing `multibody::meshcat::JointSliders`
implementation.

Resolves #17784.