RobotLocomotion/drake #17798: drakelint does not allow OpenMP use in comments

calderpg-tri, 2022-08-30:
### What happened?

We would like to include snippets like
```
#pragma omp parallel for
for (size_t index = 0; index < qs.size(); ++index) {
  operation(qs.at(i));
}
```
in documentation on using OpenMP directives with Drake code.

However, `drakelint` will incorrectly detect these uses as missing the proper `#if`/`#else` guards.

cc @rpoyner-tri @jwnimmer-tri

jwnimmer-tri, 2022-08-31:
Isn't the linter correct here, though?  If someone copy+pasted the sample code but forgot to add the guards, that would be a defect at the place where they pasted it into?

I don't understand why we want to allow for defective example code to appear?  Perhaps a specific example would be more helpful.

calderpg-tri, 2022-08-31:
The lint is right in saying that OpenMP snippets in comments should have the guards, but it is wrong on how it checks for the presence of the guards. It expects the preceding and following lines to be exactly the guards `#if defined(_OPENMP)`/`#endif`, and doesn't detect them if they are part of a comment.

> If someone copy+pasted the sample code but forgot to add the guards, that would be a defect at the place where they pasted it into?

Unguarded use in Drake _code_ would still be caught, no matter if we make the lint aware of comments or provide some nolint directive for use in comments.

jwnimmer-tri, 2022-08-31:
> ... doesn't detect them if they are part of a comment.

Can you provide a specific example?

calderpg-tri, 2022-08-31:
The lint check is basically
```
missing_pre_guard = previous_line.strip() != "#if defined(_OPENMP)"
missing_post_guard = next_line.strip() != "#endif"
```
str.strip() isn't going to remove comment characters, so
```
/// Example OpenMP loop.
///
/// #if defined(_OPENMP)
/// #pragma omp parallel for
/// #endif
/// for (size_t index = 0; index < qs.size(); ++index) {
///   operation(qs.at(i));
/// }
```
will be detected as containing the `#pragma omp` directive, but the preceding and following lines will not match the check due to the leading `///` (or any other comment characters).

jwnimmer-tri, 2022-09-01:
It might be nice to fix that, but our goal is use `/** */` not `///`  in new code, so I'm not sure urgent this is.  Removing from adoption project board.

jwnimmer-tri, 2024-03-16:
Given that API comments use `/** */`, this does not seem like an important linter problem.  Once we have a PR in flight that needs this fix to pass CI, we can fix the linter in the same PR.

closed 2024-03-16 by jwnimmer-tri