RobotLocomotion/drake #17803: [doxygen] Include information for stand-alone functions

SeanCurtis-TRI, 2022-08-31:
### What happened?

This comes from [a stackoverflow question](https://stackoverflow.com/questions/73550553/how-to-find-which-header-file-to-include-from-the-documentation)

This is a very real problem. When looking for a *class* (say [ProximityProperties](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_proximity_properties.html)), the "Detailed Description" section includes `#include <drake/geometry/geometry_roles.h>`. However, the same doesn't apply for stand-alone functions, aliases, etc. For example, the page that has the *sole* documentation for [`AddContactMaterial`](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1geometry.html#a06f08c367a0ef97e7fc397a8a44d49d8) doesn't include any information for include files. That's definitely a usability issue.

It would be very helpful if the C++ documentation provided `#include` info for every symbol so the user doesn't have to search through the source code.

jwnimmer-tri, 2022-08-31:
Isn't this what an IDE is for (or `grep`, for developers who avoid IDEs)?

Did you have any kind of fix in mind for this?  I don't see any, and I'm inclined to "won't fix" this unless you have a better idea.

We are not going to be able to change what Doxygen outputs in this case.

We could imagine adding some kind of rote comment using Doxygen markup into every non-class function, but that seems like low bang for the buck.

> This is a very real problem. ... That's definitely a usability issue.

Can you elaborate?

Thousands of people write C++ code every day without having website reference documentation that spoon feeds them the include paths for code they are including.

I wouldn't object to having the website provide this, but I don't understand why it's anywhere near a priority even worth discussing on the issue board.

I would buy a usability issue like "Users should be able to unlock the power of Drake without needing to learn C++ first", but for users who choose to program in C++, I'm not sure that include paths are a real stumbling block?

SeanCurtis-TRI, 2022-09-01:
Then I'll defer to you to respond in the stackoverflow and close this issue.

I can certainly accept the premise that doxygen does what it does and we don't have the power nor inclination to change it.

jwnimmer-tri, 2022-09-01:
I searched through https://www.doxygen.nl/manual/config.html and didn't find any configuration flag that we could toggle to add this information.  As such, I think this is "can't fix".

closed 2022-09-01 by jwnimmer-tri

RussTedrake, 2022-09-03:
FWIW -- i just took a quick look, too.  https://www.doxygen.nl/manual/config.html#cfg_show_headerfile only mentions class documentation.  And https://stackoverflow.com/questions/46422495/doxygen-show-header-to-include-in-namespace-page complained about a similar issue.