RobotLocomotion/drake #17832: Support recording of contact forces in meshcat animations.

RussTedrake, 2022-09-03:
This has been on my list, and was requested in https://stackoverflow.com/questions/73531463/recording-contact-forces-in-meshcat.

There is one subtlety to understand in making it work:  
1) Animations in meshcat currently only support recordings of transforms and properties, but not creating and destroying objects. (This is true of meshcat itself, not only Drake's mechanisms of publishing to meshcat).  It's documented in the class documentation [here](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_meshcat_animation.html).
The recordings made by MeshcatVisualizer set those transforms in a MeshcatAnimation object [here](https://github.com/RobotLocomotion/drake/blob/8dff9ef0a7385c96c89d833682328807a5ebbd51/geometry/meshcat_visualizer.cc#L170).
2) Unlike MeshcatVisualizer, the Meshcat contact visualizer [adds geometry during the simulation](https://github.com/RobotLocomotion/drake/blob/8dff9ef0a7385c96c89d833682328807a5ebbd51/multibody/meshcat/point_contact_visualizer.cc#L109). The reason for this is that we don't want to create, a priori, geometry for all possible contacts... we only want to create contacts that actually occur during a simulation. Fortunately, `MeshcatAnimations` *can* set the visibility property to true or false.

So what I was intending to implement (but have not implemented yet), is to add the recording logic to `meshcat/contact_visaulizer` that is analogous to `MeshcatVisualizer`, but which adds all of the geometry to meshcat during the recording phase so it is available, then sets it to be visible true/false during the published animation.

Assigning to @joemasterjohn as component owner.

jwnimmer-tri, 2023-04-12:
As I understand it, #18433 resolved this issue?  Almost all of the contact visualizations pass `time` now, except the few tracked separately at #19142.

\CC @DamrongGuoy

DamrongGuoy, 2023-04-13:
I confirm that contact force and moment vectors are recorded in MeshcatAnimation now.

It is also true that contact patches are not recorded correctly as tacked by #19142.

I think we can close this issue now since it's more about contact forces, not contact patches.

Please re-open, otherwise.

closed 2023-04-13 by DamrongGuoy