RobotLocomotion/drake #17833: Parse Universal Scene Description (USD) format

RussTedrake, 2022-09-04:
There is considerable momentum behind Pixar's USD format: [it's used it omniverse](https://developer.nvidia.com/usd), and you can even view in [Preview on mac](https://support.apple.com/en-me/guide/preview/prvwd129f236/11.0/mac/12.0). It should be sufficient for parsing into multibodyplant / scenegraph, and even [for rendering](https://github.com/RobotLocomotion/drake/issues/15918).

Let's use this issue so that we can track progress / intentions towards properly supporting it.

jwnimmer-tri, 2022-09-19:
FYI https://github.com/gazebosim/gz-usd might be interesting here as a stop-gap.  It offers a `usd2sdf` command-line tool.  (I haven't tried out how well it works.)

For adding Drake support for USD directly, my guess it that long-term we'd want use a native parser instead of a conversion tool, but in case users have usd files already that tool might help bridge the gap.

jwnimmer-tri, 2024-02-07:
Presuming that Drake would use https://github.com/PixarAnimationStudios/OpenUSD to parse `*.usd` files, one question will be the build system side of that library and its dependencies.  I've filed #20898 to help collate information on that front.  We can keep this issue focused on the big-picture feature.

jwnimmer-tri, 2024-02-07:
In terms of the USD parser itself, a good starting point would be to look at the [existing parsers](https://github.com/RobotLocomotion/drake/tree/master/multibody/parsing/).  The TLDR is that they all implement `class ParserInterface`, with `ParserInterface::AddAllModels` being the primary entry point.  When given a filename to parse, it's the job of that function to make API calls on [class MultibodyPlant](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html) and [class SceneGraph](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_scene_graph.html) to populate the scene.

jwnimmer-tri, 2024-10-10:
FYI Drake's SceneGraph now supports in-memory meshfiles, so the USD parser no longer needs to create a tmpdir to store meshdata.  It can load the files directly into SceneGraph without writing out the files to disk.

nepfaff, 2025-07-07:
Is this still being worked on?

jwnimmer-tri, 2025-07-07:
Not actively, no.  NVidia had been working on contributing the implementation, but they stopped.

nepfaff, 2025-07-07:
the `manipulation` repo can now make arbitrary `.sdf` files compatible with Drake, enabling the USD to SDF with `gz-usd` workflow
https://github.com/RussTedrake/manipulation/pull/453

rpoyner-tri, 2026-11-11:
Unassigning myself; I'm not actively working on this.