RobotLocomotion/drake #17860: MultibodyPlant::Clone() does not exist?

EricCousineau-TRI, 2022-09-07:
**Is your feature request related to a problem? Please describe.**
Working on studying some collision avoidance IK failures, involves constructing multiple copies of a plant, possibly mutating it (e.g. joint limits).

**Describe the solution you'd like**
Clone MultibodyPlant.

**Describe alternatives you've considered**
What we've been doing is reconstruct the same MultibodyPlant, including operators to mutate plant (e.g. joint limits).

**Additional context**
Here's my current workaround (will add to Anzu's `multibody_extras.h`); I believe it works:
```cc
namespace {
template <typename To, typename From>
std::unique_ptr<To> dynamic_pointer_cast(std::unique_ptr<From> from) {
  DRAKE_DEMAND(from != nullptr);
  std::unique_ptr<To> to(dynamic_cast<To*>(from.release()));
  DRAKE_DEMAND(to != nullptr);
  return to;
}
}

std::unique_ptr<drake::multibody::MultibodyPlant<double>>
Clone(const drake::multibody::MultibodyPlant<double>& plant) {
  return dynamic_pointer_cast<MultibodyPlant<double>>(
      plant.ToAutoDiffXd()->ToScalarType<double>());
}
```

\cc @calderpg-tri @sammy-tri

EricCousineau-TRI, 2022-09-07:
Ale, assigning ya per component chart.

\cc @xuchenhan-tri @sherm1 (since Ale's out)

jwnimmer-tri, 2022-09-07:
If we'd like to have a simple Clone sugar for systems, then we should add it to `framework::System` overall, not MbP as a special case.

jwnimmer-tri, 2022-09-07:
(It might make sense to do a MbP-specific Clone in case the System Clone were a lot more work, but I think both are pretty trivial -- we just commit your work-around as a class-member function, with slight adaptation to work across more scalar types.)

adeeb10abbas, 2023-01-31:
@jwnimmer-tri this is closed as of https://github.com/RobotLocomotion/drake/pull/18687, no?

jwnimmer-tri, 2023-01-31:
Ah, yes, I think so.

closed 2023-01-31 by jwnimmer-tri