RobotLocomotion/drake #17861: Meshcat data demarshalling is faster with msgpack-javascript

SeanCurtis-TRI, 2022-09-07:
Meshcat visualizer marshalls a *great deal* of information down the pipe. The browser needs to demarshall all of the data. For complex files with lots of models (objs) and textures (pngs) this can be dozens of megabytes of data. There was a specific [blog post](https://appspector.com/blog/how-to-improve-messagepack-javascript-parsing-speed-by-2-6-times) that suggests things can get much faster by changing how the decoding takes place.

The patch below, provided by an anonymous, good Samaritan, produced a 10X speed up in demarshalling with approximately 30 MB of scene data.  It would probably be good to push this (either into meshcat directly or, otherwise, in drake) to improve scene loading.

```diff
diff --git a/geometry/meshcat.html b/geometry/meshcat.html
index af3069e999..b1a54f1bc3 100644
--- a/geometry/meshcat.html
+++ b/geometry/meshcat.html
@@ -16,8 +16,46 @@
 
   <script type="text/javascript" src="meshcat.js"></script>
   <script type="text/javascript" src="stats.min.js"></script>
-  <script type="text/javascript" src="msgpack.min.js"></script>
+  <script type="text/javascript" src="https://cdn.jsdelivr.net/npm/@msgpack/msgpack@2.7.1/dist.es5+umd/msgpack.min.js"></script>
   <script>
+
+    // We must implement extension types 0x16 and 0x17. The trick to
+    // decoding them is they must be converted from littleEndian.
+    const extensionCodec = new MessagePack.ExtensionCodec();
+    // Uint32Array
+    extensionCodec.register({
+      type: 0x16,
+      encode: (obj) => {
+        console.error("Uint32Array encode not implemented")
+        return null;
+      },
+      decode: (data) => {
+        const to_return = new Uint32Array(data.byteLength / 4);
+        let dataview = new DataView(data.buffer, data.byteOffset, data.byteLength);
+        for (let i = 0; i < to_return.length; i++) {
+          to_return[i] = dataview.getUint32(i * 4, true);  // true b.c. littleEndian
+        }
+        return to_return
+      },
+    });
+
+    // Float32Array
+    extensionCodec.register({
+      type: 0x17,
+      encode: (obj) => {
+        console.error("Float32Array encode not implemented")
+        return null;
+      },
+      decode: (data) => {
+        const to_return = new Float32Array(data.byteLength / 4);
+        let dataview = new DataView(data.buffer, data.byteOffset, data.byteLength);
+        for (let i = 0; i < to_return.length; i++) {
+          to_return[i] = dataview.getFloat32(i * 4, true);  // true b.c. littleEndian
+        }
+        return to_return
+      },
+    });
+    
     // TODO(#16486): add tooltips to Stats to describe chart contents
     var stats = new Stats();
     var realtimeRatePanel = stats.addPanel(
@@ -49,7 +87,7 @@
     // TODO(#16486): Replace this function with more robust custom command
     //  handling in Meshcat
     function handle_message(ws_message) {
-      let decoded = msgpack.decode(new Uint8Array(ws_message.data));
+      let decoded = MessagePack.decode(new Uint8Array(ws_message.data), { extensionCodec });
       if (decoded.type == "realtime_rate") {
         rtr = decoded.rate;
       } else if (decoded.type == "show_realtime_rate") {
```

SeanCurtis-TRI, 2022-09-07:
NOTE: This patch pulls in the `msgpack.min.js` from a URL. The implementation we currently used is an explicit upstream dependency that we vendor and copy locally. We should do the same for this one.

EricCousineau-TRI, 2022-09-19:
Would love to have this in Anzu for Jammy / 22.04 transition!
esp. given removal of support for `drake-visualizer` in 22.04 (which tends to be faster than meshcat)

PR #18269 (2022-11-07): [meshcat] Update Meshcat sha (now using msgpack-javascript)
This update includes my recent meshcat PR
https://github.com/rdeits/meshcat/pull/132, which switches meshcat to using msgpack-javascript instead of msgpack-lite.  Resolves #17861.

Importantly, I've also exported the msgpack requirement via meshcat.js, so that we don't need to build it again ourselves here. msgpack-lite included the msgpack.min.js in its github distribution, but msgpack-javascript does not.  I would have needed to pull it from some CDN or NPM server, or build it locally (which would require node as a dependency).

- - [... cut]
commit 8d1a17db (2022-11-07): [meshcat] Update Meshcat sha (now using msgpack-javascript) (#18269)

This update includes my recent meshcat PR
https://github.com/rdeits/meshcat/pull/132, which switches meshcat to
using msgpack-javascript instead of msgpack-lite.  Resolves #17861.

Importantly, I've also exported the msgpack requirement via
meshcat.js, so that we don't need to build it again ourselves here.
msgpack-lite included the msgpack.min.js in its github distribution,
but msgpack-javascript does not.  I would have needed to pull it from
some CDN or NPM server, or build it locally (which would require node
as a depende [... cut]
closed 2022-11-07 by jwnimmer-tri