RobotLocomotion/drake #17862: [examples] Convert examples to use modern Drake event handling (epic)

sherm1, 2022-09-07:
Long ago we discouraged overloading of the event dispatcher [DoCalcDiscreteVariableUpdates()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_leaf_system.html#ac2d40c691d5f0335d8dcde434a8fa7b1) in favor of create Event objects with handler functions. Yet we still have an assortment of examples, tutorials, systems, and tests doing it the old way, encouraging continued bad behavior via cargo culting.

Here is what I found in Drake master. If you know of any other places that use the old method, please add them below. Fixing these will be easier with the soon-to-exist `EvalUniquePeriodicDiscreteVariableUpdate()` discussed in #10909. It is not clear whether the tests need to be updated, but if you tackle one of these please look carefully at the tests as well.

### drake/examples
- [ ] rod2d
- [ ] manipulation_station/differential_ik.py
    - [ ] manipulation_station/test/manipulation_station_test
- [ ] allegro_hand/allegro_lcm
    - [ ] allegro_hand/test/allegro_lcm_test
- [ ] planar_gripper/test/planar_manipuland_lcm_test
    - [ ] planar_gripper/test/planar_gripper_lcm_test

### drake/systems (see corresponding tests also)
- [ ] primitives/discrete_derivative
- [ ] primitives/symbolic_vector_system
- [ ] primitives/linear_system
- [ ] primitives/affine_system
- [ ] trajectory_optimization/direct_transcription

### drake/manipulation
- [ ] schunk_wsg/schunk_wsg_trajectory_generator

This came up in discussion in PR #17824 where Russ modernized DiffIK Integrator and updated the dynamic_systems tutorial in #17857.

EricCousineau-TRI, 2022-09-08:
Per component matrix + platform checklist, assigning @jwnimmer-tri

sherm1, 2022-11-09:
Notes:
- `EvalUniquePeriodicDiscreteUpdate()` was added to Drake in #18048
- WIP PR #18106 will address much of the above when merged

sherm1, 2024-05-14:
All uses of DoCalcDiscreteVariableUpdates() in Drake are gone now so this issue has outlived its usefulness.

closed 2024-05-14 by sherm1