RobotLocomotion/drake #17873: Should upgrade to libsdformat13

EricCousineau-TRI, 2022-09-08:
Not going to be released til end of month, but would be nice to get out in front of it.

Per VC, this came in (yay)! https://github.com/gazebosim/sdformat/pull/1112
Affected code:
https://github.com/RobotLocomotion/drake/blob/a2caba14526d6d3e8b0a8e783eb7d55ed303f813/multibody/parsing/detail_sdf_parser.cc#L548-L562

scpeters, 2022-09-13:
I am working on this in the following branch:

* https://github.com/RobotLocomotion/drake/compare/master...scpeters:drake:libsdformat13

scpeters, 2022-09-14:
> I am working on this in the following branch:
> 
> * [master...scpeters:drake:libsdformat13](https://github.com/RobotLocomotion/drake/compare/master...scpeters:drake:libsdformat13)

I've opened a draft pull request in https://github.com/RobotLocomotion/drake/pull/17913

PR #17913 (2022-09-28): workspace: Upgrade libsdformat to latest release 13.0.0
Closes #17873.
Closes #17907.

## Note: this is in draft form until the 13.0.0 stable release

Notable changes from libsdformat 12.5.0 that are relevant to Drake:

## New Features

* Default spec version is now 1.10
* Dependency on gz-math6 updated to gz-math7
* Dependency on gz-utils1 updated to gz-utils2
* Name migration from ignition -> gz reflected in header files (`gz/math/Vector3.hh`) and c++ namespaces (`gz::math::Vector3d`)
* Python bindings
* Warn by default on unrecognized elements
* Limit floating point truncation

## Bug fixes [... cut]
closed 2022-09-28 by jwnimmer-tri