RobotLocomotion/drake #17882: ScrewJoint: accept axis unit vector in constructor

scpeters, 2022-09-09:
**Is your feature request related to a problem? Please describe.**

The parameters to the `ScrewJoint` constructor are not consistent with the parameters used in t

The `RevoluteJoint` and `PrismaticJoint` constructors accept a unit vector parameter for the axis direction, but the `ScrewJoint` constructor has no such parameter. See the comment thread starting with https://github.com/RobotLocomotion/drake/issues/17765#issuecomment-1234928546.

**Describe the solution you'd like**

Add a `const Vector3<double>& axis` parameter to the `ScrewJoint` constructor and update the ScrewJoint mobilizer to support it to match the behavior of other joints.

**Describe alternatives you've considered**

**Additional context**

sherm1, 2022-09-11:
cc @amcastro-tri

scpeters, 2022-09-22:
I started work on this before noticing it is assigned to @amcastro-tri. I've started by adding a `Vector3<double>` argument to the joint and mobilizer constructor API in https://github.com/scpeters/drake/commit/0fea93a07a645da2e0803134600f55f34a334770, but I haven't yet updated the math from z-axis to use the arbitrary axis.

xuchenhan-tri, 2022-09-22:
@scpeters Feel free to finish the working branch and turn it into a PR to Drake. I believe the issue was assigned to Alejandro simply according to component ownership: https://drake.mit.edu/issues.html.

scpeters, 2022-10-01:
I think I've updated the math correctly, so I've opened a draft pull request in #18019

PR #18019 (2022-10-17): ScrewJoint: accept axis unit vector in constructor
Closes #17882.

## New Features

* Adds a constructor to `ScrewJoint` that accepts a `Vector3<double>` for the axis unit vector and a `screw_axis()` accessor method to the class. The current constructor that assumes the axis unit vector is the Z-axis is retained but may be considered for deprecation in the future. I adapted the documentation based on the `revolute_joint.h` header file.
* Adds a `Vector3<double>` argument for the axis unit vector to the `ScrewMobilizer` constructor and updates the mobilizer math in [34a2dab](https://github.com/ [... cut]
closed 2022-10-17 by rpoyner-tri