RobotLocomotion/drake #17889: [meshcat] Meshcat reconnected, but still says "No connection to server. Attempting to reconnect..."

RussTedrake, 2022-09-12:
### What happened?

I've been loving the new reconnect feature on Meshcat!

This time, I happened to leave Meshcat open but disconnected all night; when I ran code which made a new meshcat server, it still reconnected beautifully.  But the gui banner still says "No connection to server. Attempting to reconnect..." in the bright yellow strip.

I haven't read the code yet, but assume that having *many* failed reconnects was the difference between my experience this morning and normal?  Otherwise it could potentially just be a very low probability event with message timing?

Restarting the server yet again continued to reconnect then disappear.

### What operating system are you using?

macOS 11 (Big Sur)

### What installation option are you using?

compiled from source code using Bazel

SeanCurtis-TRI, 2022-09-12:
Presumably the browser is....safari?

Generally, this seems surprising. There are no counters (just a perpetual attempt to reconnect). A reconnection should come with a page reload. The logic is simple:

```
1. Load fresh page (message has css property `display: none`).
2. Make connection
3. Lose connection
    a. Display message (`display: block`)
    b. while True
        i. Schedule attempt to reconnect
        ii. If we reconnect, reload the page and go back to 1.
```
So, the message always starts off and only turns on with the initial disconnect. The only way to turn the message off is to reload the page. The implication of the described behavior is that somehow, your browser is hitting 3.b.ii (a successful reconnect), but it isn't doing the page reload. Therefore the successful reconnect isn't serving as a "reboot signal", it's simply reconnecting and consuming messages. That's not overtly harmful except for two things:

1. The "disconnected" message doesn't go away.
2. Depending on what you're *now* visualizing (versus what was previously being visualized), you may get undesirable artifacts malingering. It's those kind of artifacts that led to the simple expedient of reloading.

One thing we *could* do to mitigate this is to change the logic in 3.b.ii. Instead of *just* triggering the reload, we could change the display text/warning to something indicating that a temporary connection has been made and we're attempting to reload. That's better than *lying* about being disconnected but doesn't actual find the root cause. (And we'd want to make sure that if we subsequently lose connection, that the message goes back to "connection lost".)

RussTedrake, 2022-09-12:
the browser was Chrome.  sorry i forgot to specify.

jwnimmer-tri, 2023-01-23:
One thing I've seen is that with very large scenes (lots of meshes), after restarting the server the "Attempting to reconnect..." panel stays there for ~10+ seconds with no updates, seemingly until the new meshes have already been loaded via the new server.  (This is on Firefox.)  Then everything shows up pretty quickly and the yellow panel disappears.

That seems like it might be a somewhat different issue than the above, but I thought I'd toss it in here in case it helps.