RobotLocomotion/drake #17898: [pydrake] invalid value encountered in matmul

RussTedrake, 2022-09-13:
### What happened?

I've just started seeing new warning messages spewed to the console when multiplying a symbolic variable using @ in python.
The following code complains:
```
from pydrake.all import MakeVectorVariable
v = MakeVectorVariable(2, "v")
np.eye(2) @ v
```
with 
```
RuntimeWarning: invalid value encountered in matmul
  np.eye(2) @ v
array([<Expression "v(0)">, <Expression "v(1)">], dtype=object)
```

I also get it with (some) other operations:
```
from pydrake.all import MakeVectorVariable
v = MakeVectorVariable(2, "v")
np.array([1.,2.] - v)
```
causes
```
RuntimeWarning: invalid value encountered in subtract
  np.array([1.,2.] - v)
```

This is on mac with python3.10, numpy 1.21.6

### What operating system are you using?

macOS 11 (Big Sur)

### What installation option are you using?

compiled from source code using Bazel

RussTedrake, 2022-09-27:
This also happens on focal with numpy version 1.21.1

Making this high priority.  It is spamming everyone and confusing all of the students.

EricCousineau-TRI, 2022-09-28:
Re: spamming, I think we can simply suppress the warning.
It won't close the issue, but can help the UX in that students won't pay attention to it.

EricCousineau-TRI, 2022-09-28:
Additional alternative is to pin `numpy` to be at least `<1.21.6`; this will be the quickest, albeit least elegant, solution.

RussTedrake, 2022-09-28:
I'm ok with either approach in the short term. Can we provide a sufficiently narrow suppression?

EricCousineau-TRI, 2022-09-30:
I'm not sure if we can make it super narrow easily, but I'd say cost-benefit, at least from what you've mentioned per students' distraction, is that false negative is better than true positive w/ no recourse.

Confirmed repro in 0b07977dfb

EricCousineau-TRI, 2022-09-30:
Was trying to traceback versions; from 4adc2f89f6. Note that build occurs w/ host Ubuntu 20.04, `numpy==1.17.4`, so ABI issues may crop up:
- NumPy 1.16 - 1.21 - this warning appears
- NumPy <1.15 - forgot what the case was here, but `TypeError: Object arrays are not currently supported`

So I'm not sure how this was uncaught / unseen for such a long time since 1.17.4, but I think the suggested solution route is the same:
- Remove distractions, then demote priority
- Find root-cause, then fix-forward

EricCousineau-TRI, 2022-09-30:
Distraction removal wip in d533aca702

EricCousineau-TRI, 2022-10-11:
Lowering priority for now given hotfix.

rpoyner-tri, 2026-11-11:
I can no longer reproduce this. With the Drake 1.58.0 release wheel and NumPy 2.5.4 (Python 3.12, Linux), `np.eye(2) @ v`, `np.array([1.,2.]) - v` and `np.array([1.,2.] - v)` (with `v = MakeVectorVariable(2, "v")`) all return the correct result, and no `RuntimeWarning` appears even with warnings set to "always".

The warning was only seen with NumPy 1.16 to 1.21, so it looks gone with modern NumPy. Closing; please reopen if you still see it on a current NumPy.

rpoyner-tri, 2026-11-11:
Correction to my earlier comment: that test was invalid, and I'm reopening this.

I set `warnings.simplefilter("always")` before `import pydrake`. Importing pydrake installs an `ignore` filter for "invalid value encountered in" (the #18017 hotfix) that goes ahead of mine, so the warning was hidden and I wrongly concluded it was gone.

With the filter set after the import (Drake 1.58.0 release wheel, NumPy 2.5.4, Python 3.12, Linux), the warnings still fire:

```
RuntimeWarning: invalid value encountered in matmul     (np.eye(2) @ v)
RuntimeWarning: invalid value encountered in subtract   (np.array([1.,2.]) - v)
```

So the underlying cause is not fixed. Drake only hides the warning by default, and it still shows up when a user's own warning filters are applied after the import.