RobotLocomotion/drake #17909: macOS m1 CI snopt-mosek-packaging is not Uploading Artifacts

svenevs, 2022-09-13:
Relates: #17869 CC @jwnimmer-tri 

Packaging job appears to have [compiled everything successfully](https://drake-jenkins.csail.mit.edu/view/Mac%20M1/job/mac-m1-monterey-unprovisioned-clang-bazel-experimental-snopt-mosek-packaging/1/console), **however** the job reported success even though no `.tar.gz` archive was uploaded.  Likely a line or two fix in drake-ci.

mwoehlke-kitware, 2022-09-13:
Curious; it looks like CMake just went piff somewhere between L326 and L326 of `driver/configurations/bazel.cmake`? Possibly somewhere in `driver/configurations/bazel/step-create-package-archive.cmake`? (But nothing jumps out at me what could cause that...)

svenevs, 2022-09-13:
> it looks like CMake just went piff somewhere

Yeah it was a little confusing, but apparently drake-ci abuses this `PACKAGE` variable to either be a boolean or the word `"publish"`.  So I believe it creates the `.tar.gz`, just doesn't upload it and the UI doesn't make that obvious (e.g. a message saying "skipping upload" or something).  Should be fixed in https://github.com/RobotLocomotion/drake-ci/pull/181 :crossed_fingers:

jwnimmer-tri, 2022-09-13:
The tribool [meme](https://thedailywtf.com/articles/what_is_truth_0x3f_) is one of my favorite programming memes!

closed 2022-09-20 by svenevs