RobotLocomotion/drake #17910: uwebsockets needs update to latest

mwoehlke-kitware, 2022-09-13:
uwebsockets needs to be updated to the latest release. There is a build issue (details to be added).

jwnimmer-tri, 2022-09-15:
From a Drake policy perspective, it's a slight preference to use tagged releases when possible, but failing that it's a-ok to use a random git sha.

mwoehlke-kitware, 2022-09-16:
The update is also apparently causing *san failures. Will need someone to do more digging.

jwnimmer-tri, 2022-11-03:
FYI

> Note also that new_release.py is not currently working on usockets due to https://github.com/sigmavirus24/github3.py/issues/1105.

As of today, the latest usockets & uwebsockets do have an author (vs being anonymous), so that bug is no longer in play.

jwnimmer-tri, 2022-11-03:
> It looks like the latest µWebSockets may depend on an unreleased µSockets?

https://github.com/uNetworking/uSockets/issues/188

jwnimmer-tri, 2022-11-09:
See #18289.

RussTedrake, 2022-11-10:
I've just tested the new nightlies by running the first cells of "authoring multibody simulations" tutorial on a fork of the tutorials on deepnote, and confirm that connecting to meshcat:
- jammy-20221107 passed
- jammy-20221108 failed with bad gateway
- jammy-20221109 failed with bad gateway
- jammy-20221110 passed

So i do think that upgrade must have been the problem.

jwnimmer-tri, 2023-02-09:
Re-opened via the revert https://github.com/RobotLocomotion/drake/pull/18755.

Reminder of: https://github.com/RobotLocomotion/drake/issues/17910#issuecomment-1309492772.  Next step is a regression test, not an upgade.

PR #17926 (2022-09-15): [workspace] Upgrade uwebsockets to latest release v20.16.0
Closes #17910.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17926)
PR #18259 (2022-11-07): [workspace] Upgrade usockets and uwebsockets
Closes #17910.

See also https://github.com/uNetworking/uWebSockets/issues/1512.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18259)
PR #18516 (2023-01-06): Upgrade Externals January 2023
Towards #18510. Fixes #17910.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18516)