RobotLocomotion/drake #17915: macOS m1 CI failure in solvers/rotation_constraint_test

svenevs, 2022-09-14:
Relates #17869, CC @jwnimmer-tri and via blame @RussTedrake 

https://github.com/RobotLocomotion/drake/blob/7cf6c542f0779eadd72eb2b608143d0b3a057323/solvers/test/rotation_constraint_test.cc#L90

```
[12:17:02 PM]  [ RUN      ] RotationTest/TestRpyLimitsFixture.TestRpyLimits/3
[12:17:02 PM]  solvers/test/rotation_constraint_test.cc:90: Failure
[12:17:02 PM]  Expected: (x(i)) <= (ub(i)), actual: 1 vs 1
[12:17:02 PM]  solvers/test/rotation_constraint_test.cc:90: Failure
[12:17:02 PM]  Expected: (x(i)) <= (ub(i)), actual: 1 vs 1
[12:17:02 PM]  solvers/test/rotation_constraint_test.cc:90: Failure
[12:17:02 PM]  Expected: (x(i)) <= (ub(i)), actual: 1 vs 1
[12:17:02 PM]  solvers/test/rotation_constraint_test.cc:90: Failure
[12:17:02 PM]  Expected: (x(i)) <= (ub(i)), actual: 1 vs 1
[12:17:02 PM]  [  FAILED  ] RotationTest/TestRpyLimitsFixture.TestRpyLimits/3, where GetParam() = 8 (12 ms)
```

Full [build log here](https://drake-jenkins.csail.mit.edu/job/mac-m1-monterey-provisioned-clang-bazel-experimental-everything-debug/2/consoleFull), nothing obvious sticks out to me but this is preventing `@drake-jenkins-bot mac-m1-monterey-provisioned-clang-bazel-experimental-everything-debug please.` from success.  1 vs 1 not satisfying less than or equal is ... strange.

svenevs, 2022-09-28:
The test that is failing:

https://github.com/RobotLocomotion/drake/blob/e20e8e60763a3989e241af5f11a36f93f80022b1/solvers/test/rotation_constraint_test.cc#L90

I added some extra prints

```
Expected: (x(i)) <= (ub(i)), actual: 1 vs 1
x:
-3.0616169978683843e-17
 -5.302876193624534e-17
                      1
     1.0000000000000002
-9.4368957093138306e-16
 3.0616169978683793e-17
 9.4368957093138306e-16
     1.0000000000000002
  5.302876193624537e-17
ub:
1
1
1
1
1
1
1
1
1
```

This is a numerical stability problem, 1 vs 1.0000000000000002.

```diff
--- a/solvers/test/rotation_constraint_test.cc
+++ b/solvers/test/rotation_constraint_test.cc
@@ -86,8 +86,9 @@ TEST_P(TestRpyLimitsFixture, TestRpyLimits) {
             const Eigen::VectorXd& lb = b.evaluator()->lower_bound();
             const Eigen::VectorXd& ub = b.evaluator()->upper_bound();
             for (int i = 0; i < x.size(); i++) {
-              EXPECT_GE(x(i), lb(i));
-              EXPECT_LE(x(i), ub(i));
+              constexpr double epsilon = 1e-15;
+              EXPECT_GE(x(i) + epsilon, lb(i));
+              EXPECT_LE(x(i) - epsilon, ub(i)) << "x:\n" << x << "\nub:\n" << ub << '\n';
             }
           }
         }
```

Excluding my debug print after the `LE`, is this a viable solution?  This is the last m1 everything debug issue IIRC.

jwnimmer-tri, 2022-09-28:
\CC @RussTedrake and/or @hongkai-dai for feedback on the proposed tolerance change.

To my eye, the invariant of being strictly less than 1.0 (i.e., an already-feasible guess) might be important, and we need to dig further?

RussTedrake, 2022-09-28:
I think the error is clearly within the solver tolerance, and that it's fine to loosen the tolerance of this test.

PR #17998 (2022-09-30): Add tolerance to rotation_constraint_test TestRpyLimits
Fixes #17915.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17998)
commit 2677a849 (2022-09-30): Add tolerance to rotation_constraint_test TestRpyLimits (#17998)

Fixes #17915.
closed 2022-09-30 by jwnimmer-tri