RobotLocomotion/drake #17916: [geometry] Clean up drake::geometry::internal::hydroelastic

DamrongGuoy, 2022-09-14:
The file https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity/hydroelastic_internal.h has code in `drake::geometry::internal::hydroelastic` namespace that use the word `Soft` in many places.

Please change the word `Soft` to `Compliant`.

It is `internal` code, so it shouldn't affect public APIs and probably requires no deprecations.

xuchenhan-tri, 2023-08-30:
Even though the use of the word "soft" is in the internal namespace, the effect can sometimes be observed publicly. I got a question today from a confused user about the throw message

`Cannot create soft Sphere; the ('hydroelastic', 'resolution_hint') property must be positive`

caused by our choice of language [here](https://github.com/RobotLocomotion/drake/blob/189f7345adcf44c08563b1bd22a05feeb1dcde5b/geometry/proximity/hydroelastic_internal.cc#L291).

jwnimmer-tri, 2024-11-02:
Duplicate of #17147.

closed 2024-11-02 by jwnimmer-tri