RobotLocomotion/drake #17923: Add 1 Jammy pre-merge build

jwnimmer-tri, 2022-09-15:
For pull requests ("pre-merge") we currently only run Focal jobs.  We should also run one Jammy job to quickly check for any platform-specific regressions.

Please choose the fastest one (probably vanilla "release") and promote to pre-merge.

BetsyMcPhail, 2022-09-20:
Jenkins changes are ready to be applied (drake-jenkins-jobs PR [#33](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/33)). @jwnimmer-tri let me know when would be a good time to coordinate deploying on Jenkins with updating Github.

jwnimmer-tri, 2022-09-20:
A few things...

(1) Only just recently, I've realized that GCC 11 and Clang 12 have novel differences for C++20.  So actually I think the pre-merge build needs to be GCC, not Clang.  (We already test Clang 12 on Focal, though admittedly with a different libstdc++.) Sorry!

(2) Maybe we should withdraw one of the existing PR builds to balance our costs?  I think we can drop `linux-focal-gcc-bazel-experimental-release` from pre-merge, using jammy-gcc instead.  We will still have `focal-gcc-everything` in pre-merge.

(3) I think we can deploy anytime today (there's nothing "emergency" in the PR pipeline).  If you need me to click any settings on GitHub, you can just message me in `#buildcop`.  I don't have any meetings.

BetsyMcPhail, 2022-09-20:
1 and 2 sound reasonable to me and have been deployed. 

slack conversation is [here](https://drakedevelopers.slack.com/archives/C270MN28G/p1663694503110349)

BetsyMcPhail, 2022-09-20:
Github has also been updated.

closed 2022-09-20 by BetsyMcPhail