RobotLocomotion/drake #17924: More discoverable pip install version caveat

jwnimmer-tri, 2022-09-15:
As documented at https://drake.mit.edu/pip.html, prior to `pip install drake` on Ubuntu Focal 20.04, the user needs to upgrade their pip.  If they forget to do this, the error message is very weird like "no candidate version for this platform".

Some users might not even find our `pip.html` page with those instructions.

For those users, perhaps if our pypi page https://pypi.org/project/drake/ overview had some mention of this problem, more users would find out about the requirement.

Maybe we want a section at https://drake.mit.edu/troubleshooting.html about this kind of problem, and then the pypi overview page could hyperlink to that and/or pip.html?

jwnimmer-tri, 2023-01-27:
In #17960, we added a `pip.html` link to our [PyPI homepage](https://pypi.org/project/drake/).

The remaining call to action here is to write up a [troubleshooting](https://drake.mit.edu/troubleshooting.html) entry that shows the error message when `pip` is too old, along with how to fix it.

BetsyMcPhail, 2023-01-31:
@mwoehlke-kitware putting this one on your list as well.

PR #18722 (2023-02-02): Add pip troubleshooting documentation
Fixes #17924.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18722)
closed 2023-02-02 by svenevs