RobotLocomotion/drake #17934: `Trajectory` should state what times can be passed / how they are clamped?

EricCousineau-TRI, 2022-09-16:
### What happened?

In Anzu PR 9491, I was surprised that our usage of a TOTP (Time-Optimal Trajectory Parameterizatoin) primitive allowed the final desired configuration to be unequal to the last knot point in the trajectory.
This functionality was [`common_robotics_utilities::time_optimal_trajectory_parametrization::Trajectory`](https://github.com/calderpg/common_robotics_utilities/blob/5d63cb2f840cb586f1afa59315897bd3dc5fdd00/src/common_robotics_utilities/time_optimal_trajectory_parametrization.cpp#L1121).

After @calderpg-tri looked at it, he stated that it is because this trajectory class did not clamp bounds (or throw), and thus permitted extrapolation. (The fixes here are to (1) clamp in Anzu and (2) perhaps PR a fix to CRU to throw if extrapolating.)

But he saw that our `Trajectory` class *(as 1d006d66f) and its implementations had the following facets:
- `Trajectory::value()` (and other evaluators) do not impose any stated constraints on the time passed, so it could admit extrapolation.
- `Trajectory::value()` is not NVI (perhaps legacy?) and does not 
- Most implementations (if not all) saturate time to be within `[start_time, end_time]`.

Assigning ya @hongkai-dai for disposition; not sure who owns `common/trajectories` per component chart tho.

### Version

1d006d66f9993b14a509bd6bfba85a796835d22e

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2022-09-16:
It's not clear to me from the original posting up above exactly what the problem is here, or what kind of fix you're looking for?

Is it something like this documentation patch?

```patch
--- a/common/trajectories/trajectory.h
+++ b/common/trajectories/trajectory.h
@@ -30,7 +30,9 @@ class Trajectory {
 
   /**
    * Evaluates the trajectory at the given time \p t.
-   * @param t The time at which to evaluate the trajectory.
+   * @param t The time at which to evaluate the trajectory. In case it exceeds
+   * the [start_time, end_time] range, the behavior is goverened by the specific
+   * Trajectory subclass being called.
    * @return The matrix of evaluated values.
    */
   virtual MatrixX<T> value(const T& t) const = 0;
```

EricCousineau-TRI, 2022-09-16:
Not exactly that. Proposal in #17935, which is a request for unifying behaviors if appropriate.