RobotLocomotion/drake #17963: VTK Dependency Audit

svenevs, 2022-09-21:
External dependencies for the VTK builds need to be inspected.  From https://github.com/RobotLocomotion/drake/issues/16502#issuecomment-1254102689

## Ubuntu

https://github.com/RobotLocomotion/drake/blob/6f88ff94665cbf88ca87d17a4e6fe5ca443030a7/tools/workspace/vtk/image/vtk-args#L252-L274

- [ ] exprtk
- [ ] fmt
- [ ] libharu
- [ ] pugixml
- [ ] utf8

## macOS

https://github.com/RobotLocomotion/homebrew-director/blob/ed5a2b2e28eec69ee9c0aab26322c4d61a89cac2/Formula/vtk%409.1.0.rb#L345-L367

- [ ] exprtk
- [ ] fmt
- [ ] libharu

EricCousineau-TRI, 2022-09-22:
Per platform checklist, assigning @jwnimmer-tri as lead for delegation etc.

jwnimmer-tri, 2023-09-14:
Updates from f2f today:

Most likely the only remaining work is to deal with these three libraries:
- libpng
- libtiff
- libjpeg

See TODOs in `@vtk_internal`:

https://github.com/RobotLocomotion/drake/blob/d718147dd37c110004848cb37f46cae8ab58eb99/tools/workspace/vtk_internal/settings.bzl#L526-L532

https://github.com/RobotLocomotion/drake/blob/d718147dd37c110004848cb37f46cae8ab58eb99/tools/workspace/vtk_internal/settings.bzl#L550-L555

https://github.com/RobotLocomotion/drake/blob/d718147dd37c110004848cb37f46cae8ab58eb99/tools/workspace/vtk_internal/settings.bzl#L561-L566

The goal is to make sure all three of those are properly vendored, ideally with `drake_vendor_foo` prefix not just `vtkfoo`.

Remember that libtiff got angry last time we tried this.  Figure out why.

jwnimmer-tri, 2024-03-16:
`@libpng` was fixed in #20276 
`@libtiff` was fixed in #20331.
@libjpeg` was fixed in #20343.

closed 2024-03-16 by jwnimmer-tri