RobotLocomotion/drake #17979: Homebrew periodically breaks mac unprovisioned builds

ggould-tri, 2022-09-26:
Mac unprovisioned builds pull the latest homebrew; this occasionally breaks Drake by introducing a new dependency version that we do not support, or occasionally an outright version conflict.

* Errors such as missing input file 'external/vtk/include/vtk-8.2/vtk_kwiml.h' indicate that the vtk bottles need to be rebuilt because a dependency changed.
  * See, e.g., https://github.com/RobotLocomotion/drake/issues/14206
* Errors where major python libraries are missing (`ModuleNotFoundError: No module named 'scipy'` or similar) indicate that homebrew changed its python revision.
  * See, e.g., https://github.com/RobotLocomotion/drake/issues/14243
 * (please add to this list as you discover new failure modes)

This is a known issue with homebrew's release engineering philosophy and we do not expect that it will ever be fixed.  This issue exists to put it on buildcop's list of known failures and cause searches to find this.

jwnimmer-tri, 2022-09-28:
Note at least that the VTK problem will be solved by #16502 (because we'll no longer be using bottles).

jwnimmer-tri, 2023-01-18:
Closing until we have a second occurrence to note, at which point we can re-open.

closed 2023-01-18 by jwnimmer-tri