RobotLocomotion/drake #17989: Scaling for EvaluatorCost

jwnimmer-tri, 2022-09-28:
See (`class EvaluatorCost`)[https://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_evaluator_cost.html].

I think it might be nice to extend it with the capability to apply an `a` vector and `b` offset to the wrapped evaluator's outputs.

Right now, it can only wrap an evaluator with `num_outputs==1`.  We could allow for any number of outputs from the wrapped evaluator if we had `y_cost = a.dot(y_inner) + b`.  We just need to require that `a.rows() == inner_evaluator.num_outputs()`.

Relatedly, we should also bind this class into `pydrake`.

hongkai-dai, 2022-10-04:
I plan to add a new constructor
```cc
EvaluatorCost(const std::shared_ptr<EvaluatorType>& evaluator, const Eigen::Ref<const Eigen::VectorXd>& a, double b);
```
to the EvaluatorCost class.

This opens to the question whether we want to allow non-linear mapping in the future, namely
```
y_cost = g(y_inner)
```
where `g(y)` is a nonlinear function of `y`. Theoretically we can always to a EvaluatorCost as
```cc
EvaluatorCost(const shared_ptr<OuterEvaluatorType>& outer_evaluator, const shared_ptr<InnerEvaluatorType>& inner_evaluator);
```
that evaluates `outer_evaluator(inner_evaluator(x))`.

jwnimmer-tri, 2022-10-04:
For the first constructor, my vote would be to default `double b = 0.0` in the parameter declaration.  I think that in many cases we won't need an offset.

PR #18030 (2022-10-04): Construct EvaluatorCost as a linear transformation of another evaluator
Resolves #17989

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18030)
closed 2022-10-04 by jwnimmer-tri