RobotLocomotion/drake #18001: Tutorial rendering_multibody_plant broken on Deepnote

jwnimmer-tri, 2022-09-29:
Using the updated versions of #18000, the [rendering_multibody_plant.ipynb](https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/%2Frendering_multibody_plant.ipynb) on Deepnote crashes partway through.

jwnimmer-tri, 2022-09-29:
Backtrace:
```
Thread 1 "python" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140558123459008) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140558123459008) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140558123459008) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140558123459008, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007fd63d091476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007fd63d0777f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007fd5caaab1e0 in vtkXOpenGLRenderWindow::CreateAWindow() ()
   from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libvtkRenderingOpenGL2-9.1.so.1
#6  0x00007fd5caaa7687 in vtkXOpenGLRenderWindow::WindowInitialize() ()
   from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libvtkRenderingOpenGL2-9.1.so.1
#7  0x00007fd5caaac7be in vtkXOpenGLRenderWindow::Start() () from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libvtkRenderingOpenGL2-9.1.so.1
#8  0x00007fd5ca7d4835 in vtkRenderWindow::Render() () from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libvtkRenderingCore-9.1.so.1
#9  0x00007fd5caa0f29b in vtkOpenGLRenderWindow::Render() () from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libvtkRenderingOpenGL2-9.1.so.1
#10 0x00007fd5caaacb87 in vtkXOpenGLRenderWindow::Render() () from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libvtkRenderingOpenGL2-9.1.so.1
#11 0x00007fd5d00bdead in drake::geometry::render::RenderEngineVtk::PerformVtkUpdate(drake::geometry::render::RenderEngineVtk::RenderingPipeline const&) ()
   from /opt/drake/lib/python3.10/site-packages/pydrake/common/../../../../libdrake.so
```

It happens _only_ on the official tutorials machine.  Forking to a personal-account notebook doesn't repro the problem.

jwnimmer-tri, 2022-09-29:
Huh.  Xvfb wasn't running.  Doing `/opt/drake/share/drake/setup/deepnote/install_xvfb` in a terminal fixed it.

Need to check why `init.ipynb` wasn't working...

jwnimmer-tri, 2022-09-29:
The root cause seems to be that the notebook(s) sometimes hit an "Execution has been canceled" error.  If it happens during the `init.ipynb`, then the X server is not started and the notebook will crash when it tries to render.

RussTedrake, 2023-03-02:
This is still happening.  Every time I start the machine, I've seen "Execution has been canceled" errors in init.ipynb, and the rendering multibody plant tutorial crashes.  (I still think it shouldn't crash in this case: #18445 )

Was there a resolution when you closed this?  As far as I can tell that tutorial is just (still) broken.

jwnimmer-tri, 2023-03-02:
Just now in the live Tutorials site, I re-ran the init noteobok and then re-ran the rendering tutorial, and it worked fine for me.

jwnimmer-tri, 2024-03-20:
If we have new reports of not working, then we can re-open.  Seems OK for now, though.

closed 2024-03-20 by jwnimmer-tri