RobotLocomotion/drake #18016: Ubuntu Provisioned to Bazel 5.3

jwnimmer-tri, 2022-09-30:
Per #18004, we should refresh the Ubuntu Provisioned images to use Bazel 5.3.1.

Since we have test coverage of 5.3.1 via Unprovisioned already, there is no particular hurry here -- in case it makes sense to wait for any other updates to accumulate first.

jwnimmer-tri, 2022-10-24:
I'm also going to toss #18166 into this upgrade ticket (and thus, bump the +(priority: high)).  In addition to upgrading bazel, we should install `python3-opencv` into the Focal and Jammy provisioned images.

svenevs, 2022-10-26:
Focal and Jammy AMIs fully updated / upgraded and `python3-opencv` has been added.

- [sample jammy build](https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-release/699/console) `BAZEL_VERSION                  = 5.3.1`
- [sample focal build](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-bazel-continuous-release/3379/console) `BAZEL_VERSION                  = 5.3.1`

BetsyMcPhail, 2022-11-03:
The new AMIs have been up for about a week with new issues. I removed the oldest images from AWS.

closed 2022-11-03 by BetsyMcPhail