RobotLocomotion/drake #18024: Lack of Typecasting for SolverOptions / Counterintuitive Error Message

cohnt, 2022-10-03:
### What happened?

When solving a mathematical program and specifying solver options, it appears that no typecasting is performed on the solver options. If I pass in an int when a float is expected, pydrake throws a runtime error, stating that the parameter is unknown. Ideally in this situation, pydrake would cast the int to a float. Regardless of whether or not there should be this sort of automatic typecasting, it would make sense for the error message to state that the parameter was of incorrect type, as opposed to stating that the parameter itself is unknown.

### Version

45971d6c0c

### Relevant log output

```shell
>>> import pydrake.solvers
>>> prog = pydrake.solvers.MathematicalProgram()
>>> solver = pydrake.solvers.GurobiSolver()
>>> options1 = pydrake.solvers.SolverOptions()
>>> options2 = pydrake.solvers.SolverOptions()
>>> options1.SetOption(solver.id(), "TimeLimit", 30)
>>> options2.SetOption(solver.id(), "TimeLimit", 30.0)
>>> solver.Solve(prog, solver_options=options1)
Set parameter Username
Academic license - for non-commercial use only - expires 2023-09-07
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: GurobiSolver(): 'TimeLimit' is an unknown parameter in Gurobi, check https://www.gurobi.com/documentation/9.5/refman/parameters.html for allowable parameters
>>> solver.Solve(prog, solver_options=options2)
<pydrake.solvers.MathematicalProgramResult object at 0x7f56a23c7e30>
```

jwnimmer-tri, 2022-10-03:
Just to double-check my thinking here... I think the _only_ automatic conversion we would want is from `int` to `float`.  For example, we should not auto-convert from `float` to `int` even if the value had no fractional component.  Is that fair?

As part of this fix, I'll also make sure that the error message indicates the type of value we were trying to use.

cohnt, 2022-10-03:
That's the only automatic conversion I can think of. Specifying the expected value in the error message would also be a helpful change in general.

PR #18124 (2022-10-18): [solvers] Gurobi options are handled more gracefully
When an `int` is given for a `double` option, auto-promote to `double`.
When an option is given with the wrong type, explain the mismatch.

Closes #18024.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18124)
closed 2022-10-18 by xuchenhan-tri