RobotLocomotion/drake #18033: gz_* need update to latest

mwoehlke-kitware, 2022-10-04:
gz_math and gz_utils need to be updated to the latest upstream versions. Both are experiencing build errors after update.

scpeters, 2022-10-04:
I believe the `gz_math` and `gz_utils` tags are already at the latest values:

* https://github.com/gazebosim/gz-math/tags latest release number is 7.0.2
  * https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/gz_math_internal/repository.bzl#L11
* https://github.com/gazebosim/gz-utils/tags latest release number is 2.0.0
  * https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/gz_utils_internal/repository.bzl#L11

I believe the issue is that the tag formatting changed between the most recent major versions:

* https://github.com/gazebosim/gz-math/releases/tag/gz-math7_7.0.2
* https://github.com/gazebosim/gz-math/releases/tag/ignition-math6_6.13.0
* https://github.com/gazebosim/gz-utils/releases/tag/gz-utils2_2.0.0
* https://github.com/gazebosim/gz-utils/releases/tag/ignition-utils1_1.4.0

but new_release thinks the `ignition-*` tags are newer even though they have lower numbers:

~~~
gz_math_internal needs upgrade from gz-math7_7.0.2 to ignition-math6_6.13.0
gz_utils_internal needs upgrade from gz-utils2_2.0.0 to ignition-utils1_1.4.0
~~~

jwnimmer-tri, 2022-10-04:
Thanks!  I'll update our version checker to deal.

PR #18036 (2022-10-05): [workspace] Ignore old ignition tags when checking for a new_release
Somehow, the GitHub "newest first" iterator on the tags of a moved repository list the _old_ repo's tags first (from newest to oldest) followed by the _new_ repo's tags (from newest to oldest).  So, we need to filter the `gz` repositories' tags to start with `gz` in order to skip over the old tags.

I've manually tested with this PR that demoting our `gz-math` pin to the prior patch version (7.0.1) causes the tool to find 7.0.2, but running with 7.0.2 says "no upgrade required".

Closes #18033.

---
This change is [[image]](h [... cut]
closed 2022-10-05 by xuchenhan-tri