RobotLocomotion/drake #18054: RgbdSensor / DepthImageToPointCloud outputs are numerically different on two different focal machines

RussTedrake, 2022-10-07:
### What happened?

Running the following code (which uses the resources in https://github.com/RussTedrake/manipulation)
```
from pydrake.multibody.parsing import (Parser, ProcessModelDirectives,
                                       LoadModelDirectives)
from pydrake.multibody.plant import AddMultibodyPlantSceneGraph
from pydrake.systems.framework import DiagramBuilder

from manipulation.scenarios import AddRgbdSensors
from manipulation.utils import FindResource, AddPackagePaths

builder = DiagramBuilder()

plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step=0.0)
parser = Parser(plant)
AddPackagePaths(parser)
ProcessModelDirectives(
    LoadModelDirectives(FindResource("models/mustard_w_cameras.yaml")),
    plant, parser)

plant.Finalize()

AddRgbdSensors(builder, plant, scene_graph)

diagram = builder.Build()

context = diagram.CreateDefaultContext()

i=0
cloud = diagram.GetOutputPort(f"camera{i}_point_cloud").Eval(context)
cropped = cloud.Crop(lower_xyz=[-.3, -.3, -.3], upper_xyz=[.3, .3, .3])
print(cropped.xyzs()[:,:5])
```
results in the output
```
[[ 0.0040668   0.00641429  0.00740337  0.00811082  0.00855497]
 [-0.00814378 -0.00659063 -0.00543211 -0.00435735 -0.00335972]
 [ 0.19107792  0.19112574  0.19114822  0.19116543  0.19117773]]
```
on my local focal Puget, but on Deepnote running the same code gives
```
[[ 0.00405464  0.00641373  0.00740221  0.00811082  0.00855437]
 [-0.00814727 -0.00659078 -0.00543244 -0.00435735 -0.00335988]
 [ 0.19107768  0.19112572  0.19114819  0.19116543  0.19117773]]
```
deepnote link: https://deepnote.com/workspace/russtedrake-b58d8141-6691-43e8-ad76-0575b319c8b8/project/Point-Cloud-numerics-e9196296-b651-405b-8f3b-d53ddd7363e8/%2Fgrasp_candidate.ipynb 

Is that to be expected?

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

SeanCurtis-TRI, 2022-10-07:
The most obvious question is: what do the rendered images look like? Capturing the images and diffing them could do a lot to tell us the origin of the difference. (I'm alluding to the possibility of video drivers or video hardware.)

Also, what is your depth range and your clipping range for the camera. I'll be curious to see what level of precision should be expected at that particular depth (which might also be a function of video hardware/driver -- how many bits are in its depth buffer?)

SeanCurtis-TRI, 2022-10-07:
Curse my ignorance, but how does one gain access to the `models/mustard_w_cameras.yaml` file?

jwnimmer-tri, 2022-10-07:
https://github.com/RussTedrake/manipulation/blob/master/manipulation/models/mustard_w_cameras.yaml

RussTedrake, 2022-10-12:
link to slack thread (for those with access): https://drakedevelopers.slack.com/archives/C43KX47A9/p1665106826441959

RussTedrake, 2022-10-12:
tldr; it's a reasonable amount of error.. with the biggest difference likely coming from the opengl implementation on my native machine vs the opengl implementation in xvfb, which is used on the headless cloud machines.

closed 2022-10-12 by RussTedrake