RobotLocomotion/drake #18055: Mujoco XML parsing should use the diagnostic policy mechanism for errors and warnings

rpoyner-tri, 2022-10-07:
**Is your feature request related to a problem? Please describe.**
The mujoco parser does not yet use the diagnostic policy mechanism: errors are all hardcoded throw and warnings are all hardcoded to text logging.

**Describe the solution you'd like**
Use the diagnostic policy mechanism to issue Errors and Warnings. See the examples of URDF and SDFormat parsers for detail. Also, add testing to verify that making errors non-fatal will not crash the parser.

**Describe alternatives you've considered**
There are no alternatives, but limping along with the existing implementation is not strictly wrong.

**Additional context**
See also similar complaint about model directives at #18052.

rpoyner-tri, 2023-02-13:
There is subsequent work to do, The mujoco parser should provide some warning that the result of parsing w.r.t. collision filters may be surprising. See https://github.com/RobotLocomotion/drake/pull/18351#pullrequestreview-1294881692 for a fuller discussion.

rpoyner-tri, 2023-02-13:
A second piece of follow-up work is to retire the peculiar added MbP method GetMutableSceneGraphPreFinalize(), introduced in #18351.

RussTedrake, 2023-02-14:
I've addressed the "mujoco parser should provide some warning" in #18351.

rpoyner-tri, 2023-04-12:
https://github.com/RobotLocomotion/drake/pull/19157 addresses most of it, leaving a few delegated throws where MbP has nicer messages. I left a TODO for those, but I think the bulk of this issue is done and can close.

closed 2023-04-12 by rpoyner-tri